Ysurac/openmptcprouter #188: Add a way to auto test best tcp congestion control

Ysurac, 2018-10-12:
BBR on both side is not always the best choice.

its27, 2018-10-12:
Do you know in what cases BBR is not best?

Ysurac, 2018-10-12:
I have better result using bbr on VPS side and wvegas on OMR side when aggregating FTTH, 4G and xDSL. But this really need more tests.

rstanislav, 2018-10-16:
From my experience wvegas is best for wireless(3g/4g etc..) connections using MPTCP, also MPTCP dev's don't recommend using different congestions controls on client/server, they always say that they must be the same.

SoGooDFR, 2018-10-20:
I've only 4G link and i've tested wvegas for one week. For me the BBR is the best actualy.

gsxryan, 2021-06-24:
When the MPTCP setting is changed in the web server console, does it also change it on the VPS?  I've been testing a lot on 2x 1Gb FTTH connections and ended up finding a problem with mismatched port speeds on my UDM.  Need to do all this testing again soon.  Let me know if you have any tests you'd like to try with theoretical 2Gb symmetrical

BBR2 better than BBR?