Ysurac/openmptcprouter #189: OMR-Bypass protocole feature is unstable

Syam, 2018-10-12:
## Expected Behavior
Adding/Editing HTTP protocoles in OMR-Bypass should always work.

## Actual Behavior
Behavior is not consistent.
Browsing a website like http://www.myipaddress.com will sometimes display VPS address instead of my WAN1 address, or will not work at all.

## Steps to Reproduce the Problem

  1. On a fresh install, bypass/add HTTP protocole to WAN1 interface and save/apply
  2. Browse a website that show your address, you should see WAN1 address
  3. repeat CTRL+F5 will sometimes show VPS address
  4. Add another protocole like SSL to WAN1 interface and save/apply
  5. repeat CTRL+F5 will **always** show VPS address, nothing work at all.
  6. reset OMR-Bypass configs, and add HTTP protocole to WAN1 agin
  7. repeat CTRL+F5 will show random addresses

## Specifications

  - OpenMPTCProuter version: v0.37 & v0.38
  - OpenMPTCProuter VPS version: v0.37 & v0.38
  - OpenMPTCProuter platform: x86_64
  - Wan interfaces: Wan1(eth1) to ADSL + Wan2(wan2) to 4G router

Maudissure, 2018-10-13:
I hijack this issue to say that after it works, i lost Netflix bypass in a couple hour.

Is anyone else have this unstability ?

Ysurac, 2018-10-13:
I would need system log. I think I forget to apply again bypass rules after a shadowsocks restart...
This will be fixed in next release.

Maudissure, 2018-10-13:
@TheCrach  How did you do that please ?

@Ysurac The fix is in the 0.38.1 ?

TheCrach, 2018-10-13:
Edit "/bin/omr-tracker-ss" using vi or nano, I'm using 0.38.1, the issue is still there

Maudissure, 2018-10-13:
Mmmmh, still have Netflix proxy error. Is there a time to wait ?

Ysurac, 2018-10-13:
Fix will be in OpenMPTCProuter v0.38.2 that CircleCI is compiling. Should be available in a few hours.

Syam, 2018-10-13:
> Edit "/bin/omr-tracker-ss" using vi or nano, I'm using 0.38.1, the issue is still there

@TheCrach Wow thank you so much, that was the original issue i got about netflix, testing omr-bypass around with simple HTTP protocole that lead me to write this ticket.

@Ysurac you're so reactive

Ysurac, 2018-10-13:
Should be fixed (if I didn't make mistake) in 0.38.2 that is now available.

Maudissure, 2018-10-13:
Works for me.

Ysurac, 2018-10-13:
* I've made some changes on v0.38.3 that is compiling. I tested it with telnet and seems to work well. Will be available in a few hours.
* There is no html protocol, if you mean HTTP this will never work properly if Shadowsocks tracker is enabled in omr-tracker interface.
* reset button is working: it reset to default settings if you don't save.

TheCrach, 2018-10-13:
Great, I was currently trying to fix that in omr-bypass.lua, but in init.d is another solution, thank you ;)

TheCrach, 2018-10-13:
Shouldn't there be a "uci commit dhcp.dnsmasq" and "/etc/init.d/dnsmasq reload" in omr-bypass ? I'm obliged to do that to get omr-bypass working after modifications

TheCrach, 2018-10-13:
Would also be great to add -q flag on these ipset rules to avoid errors when IPs are already added and omr-bypass is reloaded:

> if [ "$valid_ip4" = "ok" ]; then
> 		ipset add ss_rules_dst_bypass_$type $ip
> 	elif [ "$valid_ip6" = "ok" ]; then
> 		ipset add ss_rules6_dst_bypass_$type $ip
> 	fi

Thank you

Ysurac, 2018-10-14:
OpenMPTCProuter v0.38.3 is compiling again with these changes applied. Should be out in a few hours. Thanks.

KoRnEr007, 2018-10-14:
Netflixis is not working after this update.
I reboot all my devices and my vps too....

TheCrach, 2018-10-14:
If you updated from previous versions and restored the config you must delete the dnsmasq ipset line in /etc/config/dhcp
We need more details btw, what are your omr-bypass settings ?

TheCrach, 2018-10-14:
Remove the two ipset lines, "/etc/init.d/omr-bypass reload" and your problem will be solved

KoRnEr007, 2018-10-14:
After the remove, now i can add my lines to add nflxvideo.net and netflix.com here http://192.168.100.1/cgi-bin/luci/admin/services/omr-bypass?

TheCrach, 2018-10-14:
Yes

KoRnEr007, 2018-10-14:
not working :( maybe i have a config missing ? SHADOWSOCKS is not activate but OMR-bypass worked withtout before.

Ysurac, 2018-10-14:
What DNS are you using from your computer ?
What is the result of ```ipset --list```, ```iptables-save | grep MARK``` and ```ip rule``` ?

KoRnEr007, 2018-10-14:
My dns on my computer is :
192.168.100.1 (dhcp)

ipset --list (of my router)

root@OpenMPTCProuter:~# ipset --list
Name: ss_rules6_src_bypass
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 2
Number of entries: 0
Members:

Name: ss_rules6_src_forward
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 2
Number of entries: 0
Members:

Name: ss_rules6_src_checkdst
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 2
Number of entries: 0
Members:

Name: ss_rules6_dst_bypass
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 5
Number of entries: 0
Members:

Name: ss_rules6_dst_bypass_all
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 8
Number of entries: 0
Members:

Name: ss_rules6_dst_bypass_
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1352
References: 3
Number of entries: 2
Members:
fe80::/10
fd00::/8

Name: ss_rules6_dst_forward
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 2
Number of entries: 0
Members:

Name: ss_rules6_dst_forward_recrst_
Type: hash:ip
Revision: 4
Header: family inet6 hashsize 64 maxelem 65536 timeout 3600
Size in memory: 104
References: 0
Number of entries: 0
Members:

Name: ss_rules_src_bypass
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 2
Number of entries: 0
Members:

Name: ss_rules_src_forward
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 2
Number of entries: 0
Members:

Name: ss_rules_src_checkdst
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 2
Number of entries: 0
Members:

Name: ss_rules_dst_bypass_all
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 8
Number of entries: 0
Members:

Name: ss_rules_dst_bypass
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 5
Number of entries: 0
Members:

Name: ss_rules_dst_bypass_
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 1560
References: 3
Number of entries: 20
Members:
192.0.2.0/24
127.0.0.0/8
203.0.113.0/24
169.254.0.0/16
100.64.0.0/10
255.255.255.255
192.88.99.0/24
192.52.193.0/24
224.0.0.0/4
0.0.0.0/8
192.168.0.0/16
192.175.48.0/24
192.31.196.0/24
198.18.0.0/15
198.51.100.0/24
51.75.126.252
172.16.0.0/12
10.0.0.0/8
192.0.0.0/24
240.0.0.0/4

Name: ss_rules_dst_forward
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 408
References: 2
Number of entries: 1
Members:
8.8.8.8

Name: ss_rules_dst_forward_recentrst_
Type: hash:ip
Revision: 4
Header: family inet hashsize 64 maxelem 65536 timeout 3600
Size in memory: 88
References: 0
Number of entries: 0
Members:

Name: ss_rules_dst_bypass_eth0
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 1
Number of entries: 0
Members:

Name: ss_rules6_dst_bypass_eth0
Type: hash:net
Revision: 6
Header: family inet6 hashsize 64 maxelem 65536
Size in memory: 1128
References: 1
Number of entries: 0
Members:

iptables-save | grep MARK
root@OpenMPTCProuter:~# iptables-save | grep MARK
-A ss_rules_pre_src -m set --match-set ss_rules_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A PREROUTING -m set --match-set ss_rules_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff
-A PREROUTING -m set --match-set ss_rules_dst_bypass_eth0 dst -j MARK --set-xmark 0x5393/0xffffffff
-A ss_rules_pre_src -m set --match-set ss_rules_dst_bypass_all dst -j MARK --set-xmark 0x539/0xffffffff

ip rule!:
root@OpenMPTCProuter:~# ip rule
0:      from all lookup local
0:      from all fwmark 0x1 lookup 100
0:      from 192.168.11.2 lookup 3
0:      from 192.168.10.2 lookup 4
0:      from 10.255.255.2 lookup 5
1:      from all fwmark 0x539 lookup 991337
1:      from all fwmark 0x5393 lookup 3
1:      from all fwmark 0x5394 lookup 4
96:     from 192.168.10.2 lookup 4
97:     from 192.168.11.2 lookup 3
100:    from all lookup lan
10000:  from 192.168.100.1 lookup lan
20000:  from all to 192.168.100.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90002:  from all iif lo lookup lan

Ysurac, 2018-10-14:
OK, I find the problem. This will be solved (really (I hope)) in next release :)

KoRnEr007, 2018-10-14:
Ok :) it's not me the prb ;)

Ysurac, 2018-10-14:
With macvlan wrong rules are created, so this is not working when an interface is selected (stupid mistake I made).
This will be fixed in 0.38.4 that is compiling and should be out in a few hours.

KoRnEr007, 2018-10-14:
i have just to update with 0.38.4 and it will be good ?

Ysurac, 2018-10-14:
When it will be out yes.

Syam, 2018-10-14:
> * There is no html protocol, if you mean HTTP this will never work properly if Shadowsocks tracker is enabled in omr-tracker interface.

Yes i mean HTTP (i updated my message)

> * reset button is working: it reset to default settings if you don't save.

Oh ok, i expected it to erase all defined rules.

> With macvlan wrong rules are created, so this is not working when an interface is selected (stupid mistake I made).

Hehe, stupid mistakes lead to great developpers !

> This will be fixed in 0.38.4 that is compiling and should be out in a few hours.

Yeah, can't wait ! I installed my 4G antenna this afternoon with greats bandwith results, going up from 12/1mbps on single ADSL to 60/16mbps through openmpctprouter ! Netflix is the last point to fix for me :)

KoRnEr007, 2018-10-14:
Syam :) your pleasure will be better

Syam, 2018-10-14:
Upgraded to v0.38.4 and Netflix still not working :(

- bypass netflix.com and nflxvideo.net domains
- bypass Netflix protocol

Lan interface: eth0
Wan interfaces: Wan1(eth1) to ADSL + Wan2(wan2->eth1) to 4G router

Maybe i need to reset all my config ?

```
root@OpenMPTCProuter:~# ipset --list
Name: ss_rules_src_bypass
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 1
Number of entries: 0
Members:

Name: ss_rules_src_forward
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 1
Number of entries: 0
Members:

Name: ss_rules_src_checkdst
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 1
Number of entries: 0
Members:

Name: ss_rules_dst_bypass_all
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 5
Number of entries: 0
Members:

Name: ss_rules_dst_bypass
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 344
References: 3
Number of entries: 0
Members:

Name: ss_rules_dst_bypass_
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 1432
References: 2
Number of entries: 20
Members:
10.0.0.0/8
192.0.0.0/24
224.0.0.0/4
169.254.0.0/16
240.0.0.0/4
[... 207 lines cut ...]
-A QOS_MARK_eth1 -m dscp --dscp 0x30 -j MARK --set-xmark 0x1/0xff
-A QOS_MARK_eth1 -m dscp --dscp 0x2e -j MARK --set-xmark 0x1/0xff
-A QOS_MARK_eth1 -m dscp --dscp 0x24 -j MARK --set-xmark 0x1/0xff
-A QOS_MARK_eth1 -m tos --tos 0x10/0x3f -j MARK --set-xmark 0x1/0xff
-A omr-bypass-dpi -m ndpi  --NetFlix  -j MARK --set-xmark 0x5395/0xffffffff
```

```
root@OpenMPTCProuter:~# ip rule
0:      from all lookup local
0:      from 10.255.255.2 lookup 3
0:      from 192.168.0.250 lookup 5
0:      from 192.168.2.250 lookup 6
1:      from all fwmark 0x539 lookup 991337
1:      from all fwmark 0x5396 lookup 6
1:      from all fwmark 0x5395 lookup 5
96:     from 192.168.2.250 lookup 6
97:     from 192.168.0.250 lookup 5
100:    from all lookup lan
10000:  from 192.168.100.1 lookup lan
20000:  from all to 192.168.100.1/24 lookup lan
32766:  from all lookup main
32767:  from all lookup default
90005:  from all iif lo lookup lan
```

Syam, 2018-10-14:
I reseted all my conf, no more chance.

KoRnEr007, 2018-10-15:
Ok for me, all is working.
Iptv + netflix OK

Thanks for your job

KoRnEr007, 2018-10-15:
if you add a new line... netflix not working :(

Ysurac, 2018-10-15:
@Syam you bypass netflix using eth1 and wan2. I think there is something strange in your configuration.
Can you give me the result of ```uci show network``` ?

Ysurac, 2018-10-15:
And the result of ```uci show omr-bypass``` ?

Syam, 2018-10-15:
```
root@OpenMPTCProuter:~# uci show omr-bypass
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.globals=globals
network.globals.ula_prefix='fd0d:2e86:80d1::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.metric='2048'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='b8:ae:ed:73:0b:97'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.@route6[0]=route6
network.@route6[0].interface='lan'
network.@route6[0].target='::/0'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.wan1=interface
network.wan1.ifname='eth1'
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.macaddr='00:24:9b:0c:3e:16'
network.wan1.netmask='255.255.255.0'
network.wan1.ipaddr='192.168.0.250'
network.wan1.gateway='192.168.0.254'
network.wan2=interface
network.wan2.proto='static'
network.wan2.type='macvlan'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth1'
network.wan2.macaddr='9a:c0:a8:58:a6:1e'
network.wan2.ipaddr='192.168.2.250'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.2.1'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth1'
network.wan2_dev.macaddr='9a:c0:a8:58:a6:1e'
```
```
root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='netflix.com'
omr-bypass.@domains[0].interface='eth1'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='nflxvideo.net'
omr-bypass.@domains[1].interface='eth1'
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='NetFlix'
omr-bypass.@dpis[0].interface='eth1'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.wan2=interface
omr-bypass.wan2.id='6'
omr-bypass.eth1=interface
omr-bypass.eth1.id='5'
```

KoRnEr007, 2018-10-15:
Syam ? did you use on DNS : 192.168.100.1 (router ip)?

Syam, 2018-10-15:
> Syam ? did you use on DNS : 192.168.100.1 (router ip)?
> Why do you have eth1 and eth0 ?
> At home, eth0 = lan wan1 and wan2

Omg, my DNS was bind directly to 8.8.8.8, an old test ...

@KoRnEr007 you saved my life, works like a charm, thank you !

closed 2018-10-15 by Syam