Ysurac/openmptcprouter #191: Suggestion: IPTV friendly option

TheCrach, 2018-10-16:
Hello,

I had a hard time finding out why my Orange IPTV packets were dropped, I finally realized that those packets do not really like to be marked by iptables (if using omr-bypass).
I added my IP list in /usr/bin/ss-rules inside "o_dst_bypass_" and I suppose it would work inside Services > Shadowsocks > Redir rules > Destination Settings
It would be more user-friendly to provide a direct option in omr-bypass or elsewhere for IPs that should not be marked and simply using ss-rules and static routes

Here is a useful list:
# Free IPTV
	212.27.32.0/19

# Orange IPTV
	80.10.117.120/31
	80.10.204.0/22
	81.253.206.0/24
	81.253.210.0/23
	81.253.214.0/23
	193.253.67.88/29
	193.253.153.227/24
	193.253.153.228/24
	172.23.12.0/22
	172.20.224.167/24
	172.19.20.0/23

> 80.10.117.120/31 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
80.10.204.0/22 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
81.253.206.0/24 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
81.253.210.0/23 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
81.253.214.0/23 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
172.19.20.0/23 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
172.20.224.167 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
172.23.12.0/22 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
193.253.67.88/29 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
193.253.153.227 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231
193.253.153.228 via 10.155.55.254 dev orangedsl proto static src 10.155.55.231

I know it is not a really common configuration but at least people knows how to fix IPTV issues with this message :)

KoRnEr007, 2018-10-16:
very interesting point.

Ysurac, 2018-10-17:
I don't see how packet marking can change anything for Orange as marking is not visible on another host (I don't change TOS). The issue must be somewhere else.

closed 2018-11-26 by Ysurac