Ysurac/openmptcprouter #194: Lost interface aggrégation

ThibautBascoulergue, 2018-10-21:
## Expected Behavior

Difficult to explain because it does not happen all the time.
Sometimes I lose one of my two interfaces at the level of aggregation while it is up at the omr tracker.
The only solution to find the aggregates is to restart the interface or reboot openmpctp

## Actual Behavior
[image]

But on MPCTP BW 👍 
[image]

## Steps to Reproduce the Problem

  1. Wait..
  2. If i restart WAN2 interface 👍 
[image]

## Specifications

  - OpenMPTCProuter version: (38.4)
  - OpenMPTCProuter VPS version: (0.55)
  - OpenMPTCProuter platform: x86_64

Ysurac, 2018-10-21:
Seems to have a problem on wan1, can you give the result of *System*->*OpenMPTCProuter*->*MPTCP support check* on both interface ?

ThibautBascoulergue, 2018-10-21:
its because SFR network...

1: 192.168.1.1 
2: * 
3: 10.4.0.1  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
4: 172.19.168.251  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 77.154.231.169  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 109.26.69.202  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
7: * 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: * 
23: * 
24: * 
25: * 
26: * 
27: * 
28: * 
29: * 
30: * 
31: * 
32: * 
33: * 
34: * 
35: * 
36: * 
37: * 
38: * 
39: * 
40: * 
41: * 
42: * 
43: * 
44: * 
45: * 
46: * 
47: * 
48: * 
49: * 
50: * 
51: * 
52: * 
53: * 
54: * 
55: * 
56: * 
57: * 
58: * 
59: * 
60: * 
61: * 
62: * 
63: * 
64: * 

Result:
nil

Ysurac, 2018-10-21:
I would need system log and the result of ```multipath``` command when you have the problem.

SoGooDFR, 2018-10-28:
I test it and we have nothing in the system log.

ThibautBascoulergue, 2018-11-03:
You can close :)

closed 2018-11-03 by Ysurac