Ysurac/openmptcprouter #198: OMRBypass(domain) and slow reboot

PxxBxx, 2018-10-28:
## Expected Behavior

Set Domains in OMR-Bypass feature, and have a nominal boot time

## Actual Behavior

When setting domains in OMR-Bypass, the reboot takes ~5 minutes

## Steps to Reproduce the Problem

  1. Set domains in OMR-Bypass and assign them wan1 (or default)
          netflix.com, netflix.net, nflxvideo.com, nflxext.com, ...
  2. Reboot
  3. wait... 5min approx

Note : when I set only NDI protocols, it's OK. I didn't try IPs, but the slow reboot occurs as soon as I set domains in OMR-Bypass. Maybe a name resolution at boot time, without an interface up ?

## Specifications
  - OpenMPTCProuter version: 0.39.2 (also 0.39.1)
  - OpenMPTCProuter VPS version: 0.55
  - OpenMPTCProuter platform: RPI3

## Additional traces

### Ping from client to monitor the reboot time
PING 192.168.100.1 (192.168.100.1): 56 data bytes 
64 bytes from 192.168.100.1: icmp_seq=0 ttl=64 time=1.127 ms
64 bytes from 192.168.100.1: icmp_seq=1 ttl=64 time=0.923 ms
64 bytes from 192.168.100.1: icmp_seq=2 ttl=64 time=1.339 ms
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
…
Request timeout for icmp_seq 308
ping: sendto: Host is down
Request timeout for icmp_seq 309
Request timeout for icmp_seq 310
64 bytes from 192.168.100.1: icmp_seq=311 ttl=64 time=1.452 ms

### Kernel log at boot time
[   12.672170] brcmfmac mmc1:0001:1: no of_node; not parsing pinctrl DT
[   12.672275] brcmfmac mmc1:0001:2: no of_node; not parsing pinctrl DT
[   12.675756] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   12.857119] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   12.875705] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[   12.939278] brcmfmac mmc1:0001:3: no of_node; not parsing pinctrl DT
[   12.939427] usbcore: registered new interface driver brcmfmac
[   12.947116] usbcore: registered new interface driver ipw
[   12.953229] usbserial: USB Serial support registered for IPWireless converter
[   12.962341] usbcore: registered new interface driver option
[   12.968706] usbserial: USB Serial support registered for GSM modem (1-port)
[   12.977729] usbcore: registered new interface driver qcserial
[   12.984530] usbserial: USB Serial support registered for Qualcomm USB modem
[   13.005004] usbcore: registered new interface driver rt2800usb
[   13.024902] usbcore: registered new interface driver rtl8192cu
[   13.032024] kmodloader: done loading kernel modules from /etc/modules.d/*
[  287.520153] 8021q: adding VLAN 0 to HW filter on device eth0
[  287.532681] 8021q: adding VLAN 0 to HW filter on device wan1
[  287.543236] 8021q: adding VLAN 0 to HW filter on device wan2
[  289.165984] wan1: hw csum failure
[  289.170311] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.14.78 #0
[  289.177370] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
[  289.184581] Call trace:
[  289.187869] [<ffffff8008086bf0>] dump_backtrace+0x0/0x3b0
[  289.194169] [<ffffff8008086fb4>] show_stack+0x14/0x20
[  289.200096] [<ffffff8008548138>] dump_stack+0x9c/0xbc
[  289.205905] [<ffffff800842b5d0>] netdev_rx_csum_fault+0x38/0x48
[  289.212590] [<ffffff800841efac>] __skb_checksum_complete+0x6c/0xc0
[  289.219559] [<ffffff8008493ae8>] tcp_v4_rcv+0x1c0/0x10c0
[  289.225619] [<ffffff800846e1c8>] ip_local_deliver_finish+0x130/0x1c8
[  289.232732] [<ffffff800846e6d8>] ip_local_deliver+0x68/0xc0
[... 47 lines cut ...]
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp: Specific info options:
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -b, --broadcast                 Display calculated broadcast address
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -m, --netmask                   Display netmask for IP
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -n, --network                   Display network address
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -p, --prefix                    Display network prefix
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --minaddr                   Display the minimum address in the network
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --maxaddr                   Display the maximum address in the network
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --addresses                 Display the maximum number of addresses in
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:                                   the network
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --addrspace                 Display the address space the network
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:                                   resides on
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -h, --hostname                  Show hostname determined via DNS
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -o, --lookup-host=STRING        Show IP as determined via DNS
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp: Other options:
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --class-prefix              When specified the default prefix will be determined
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:                                   by the IPv4 address class
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --no-decorate               Print only the requested information
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -s, --silent                    Don't ever display error messages
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -v, --version                   Display program version
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:   -?, --help                      Show this help message
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp:       --usage                     Display brief usage message
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp: /etc/rc.common: eval: line 1: valid_subnet6: not found
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp: /etc/rc.common: eval: line 1: valid_subnet6: not found
Sun Oct 28 12:47:40 2018 daemon.notice procd: /etc/rc.d/S90mptcp: ipcalc: ip address expected

Ysurac, 2018-10-28:
Log will be fixed in 0.39.3 that will be out in a few hours and boot time should be shorter.

PxxBxx, 2018-10-28:
OK, I'll try as soon as I can ! thanks for the hard work

ThibautBascoulergue, 2018-10-29:
Same problem with 39.3

ThibautBascoulergue, 2018-10-29:
slow boot time. i must wait five mins before can ping openmptcp router.

And after boot is over, no dns resolution :/

I must remove all entry in omr-bypass to recover internet connexion.

Ysurac, 2018-10-29:
Can you give me the result of ```uci show omr-bypass``` that make a slow boot time ?

Ysurac, 2018-10-29:
I found the problem. This will be solved in next release.

ThibautBascoulergue, 2018-10-29:
nice :)

ThibautBascoulergue, 2018-10-30:
you know when the next release will be compiled ?

thx

Ysurac, 2018-10-30:
Before the end of the week. Maybe tomorrow.

ThibautBascoulergue, 2018-10-30:
ok thank's

ThibautBascoulergue, 2018-11-01:
Sorry but with 0.40 version, same problem.

all work prefectly but, if i add omr-bypass domain for netflix and reboot.

the reboot take long time..

Network interface up after few minutes, see the screenshot 👍 
[image]

Ysurac, 2018-11-01:
True, I forget a little change. Should be really fixed in 0.40.1 that will be out in a few hours.

PxxBxx, 2018-11-02:
I've just installed 0.40.1 (and VPS 0.62)...
Everything is OK, reboot is taking nominal time, no waiting, no slow boot.
Thanks a lot, I'll look into all enhancements (no-delay seems promising)

ThibautBascoulergue, 2018-11-02:
Nice. Very good job. Thank's