Ysurac/openmptcprouter #199: Wich wan is use for open tcp/udp session ?

SoGooDFR, 2018-10-28:
Wich wan is use for open tcp/udp session ? Master ?

Ysurac, 2018-10-29:
The wan with the lower RTT if bandwidth of it is not already full used.

rstanislav, 2018-10-29:
I think question is what wan is used for initial connection (default gw), as there is already tracking tech for status page, maybe it will be possible to implement mechanics for default gw not first currently active, but best based on RTT ? And change it dynamically if RTT changes and difference between current default GW and GW with best RTT is more than 50% for example ?

Ysurac, 2018-10-29:
For initial connection it's the connection defined as master for multipath that is used. There is no need to change to best RTT because it's used only for initial connection of ShadowSocks and VPN, then it's connection with best RTT that is used. Connection with the better stability should be used as master.

SoGooDFR, 2018-10-29:
With 4G/ADSL agregation, we need to select which connection is used for open TCP/UDP session because the 4G has some limitation of 500 open sessions.

Ysurac, 2018-10-29:
4G have a RTT higher than ADSL so not really a problem.

rstanislav, 2018-10-29:
> For initial connection it's the connection defined as master for multipath that is used. There is no need to change to best RTT because it's used only for initial connection of ShadowSocks and VPN, [...]

Yes, it is true, but if master interface quality(RTT etc..) degrades, for example in moving vehicle, and at some point loses connection this usually cause loss of connectivity, because OMR tracker even with low timeouts not detecting this fast enough, and also, this only works for already established connections, but for new connections(user opens new web site for example) shadowsocks also opens new connections and as result starting subflow can be very slow... for example master at start have perfect signal and 4G(LTE) connectivity, but over time signal drops to 20-30% and to 3G, this affects overall user experience greatly.

Ysurac, 2018-10-30:
An option to always use connection with lower RTT as master will be available in next release.

rstanislav, 2018-11-01:
So far it works way more better, will do more testing and report result, i think this addition was biggest improvement because now allows to overcome current biggest problem of mptcp - single default route for initial connection, that was often not best in situations when used with wireless links that can degrade link quality over time!