Ysurac/openmptcprouter #200: Suggestion - Disable Nagle's Algortihim by default

joshuaeb09, 2018-10-28:
## Expected Behavior
Shadowsocks should handle smaller than full MSS packets without much delay.

## Actual Behavior
Interactive applications such as websockets or small UDP packets via shadowsocks suffer impaired performance under default configuration.  Additionally it slows down certain TLS processes noticeably.

## Steps to Reproduce the Problem
Deploy with defaults and then attempt to use highly interactive applications with small packet sizes via shadowsocks.

## Solution 
Use --no-delay to disable nagle on both server side and router for shadowsocks as nagle is typically of no or negative benefit on modern networks. 

## Specifications

  - OpenMPTCProuter version: 0.39.3
  - OpenMPTCProuter VPS version: 0.58
  - OpenMPTCProuter platform: X86

I would advise killing nagle in the default configuration as it is troublesome with most modern applications.  At work it causes numerous issues for customers and we typically kill it 99% of the time when an ADC is front-ending web-traffic as most of the lower cost ADC's do not intelligently disable it when required.  It's most noticeable impact can be seen in TLS and general web-browsing functions that pretty much every user will encounter.  This dated, but still very valid publication from NYU explains it as it would impact most end users.

https://cs.nyu.edu/artg/publications/Buff_Goldberg_Web_Servers_Should_Turn_Off_Nagle_to_Avoid_Unnecessary_200_ms_Delays_1999.pdf

joshuaeb09, 2018-11-01:
Did you run into any issues with nagle off ?  I noticed VPS .60 was out and after upgrade I had to re-patch my systemd file to re-disabled nagle on the server-side.  

< OpenMPCTProuter VPS 0.60 >

Pre-Patch :
root@mptcp-server:~# cat /lib/systemd/system/shadowsocks-libev-server@.service | grep ss-server
Documentation=man:ss-server(1)
ExecStart=/usr/bin/ss-server -c /etc/shadowsocks-libev/%i.json

root@mptcp-server:~# ps aux | grep shadow
root     26384  0.0  0.0  24668  1548 ?        Ss   00:52   0:00 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json
root     26392  0.0  0.0  24668  1412 ?        Ss   00:52   0:00 /usr/bin/ss-server -c /etc/shadowsocks-libev/config2.json
root     27940  0.0  0.0  12788  1016 pts/2    S+   00:59   0:00 grep shadow

Post Patch :
root@mptcp-server:~# cat /lib/systemd/system/shadowsocks-libev-server@.service | grep ss-se
Documentation=man:ss-server(1)
ExecStart=/usr/bin/ss-server -c /etc/shadowsocks-libev/%i.json --no-delay

root@mptcp-server:~# ps aux | grep ss-server
root       656  0.0  0.0  24668  1376 ?        Ss   01:05   0:00 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json --no-delay
root       658  0.0  0.0  24668  1408 ?        Ss   01:05   0:00 /usr/bin/ss-server -c /etc/shadowsocks-libev/config2.json --no-delay
root      1391  0.0  0.0  12788   944 pts/0    S+   01:05   0:00 grep ss-server

Ysurac, 2018-11-01:
I added it in json config but this config is not updated, it's added on command line in OpenMPTCProuter VPS script 0.61 :)

joshuaeb09, 2018-11-01:
Awesome.  I just created a new test instance and I see it there by default now.  Going to go ahead and close this.

closed 2018-11-01 by joshuaeb09