Ysurac/openmptcprouter #201: Something went wrong, WAN's with DHCP no longer working

rstanislav, 2018-10-31:
So after vacation i compiled last version from source and now WAN's with dhcp client not working - on status page only 1 always active, all other (4 in my case) is "No server ping response after 1 second".
And also if i try to do ping -I eth1...2...3 etc only 1 works, all other shows timesouts.
There is alot of pushes of code since version 0.36.4r7317-f506de2cda(image i have on sd card that is working without this problem), cant find what exactly happened and what caused this..

I'm using RPi 3B+ and 4 USB modems.

Will try to flash version from archive to find when this happened.

Also during tests i have found a bug in /etc/init.d/shadowsocks-libev
There is check that was renamed from rules_status to rules_exist, and rules_exist in missing in 
EXTRA_COMMANDS="rules_up rules_down" so this causes errors when its called from post-tracking.d/post-tracking (because it says unknown command if called from console) i think it should be EXTRA_COMMANDS="rules_up rules_down rules_exist" ?

Ysurac, 2018-10-31:
I think I found the issue, in ```init.d/mptcp``` the check ```[ "$( valid_subnet4 $gateway )" = "ok" ]``` should be ```[ "$( valid_subnet4 $gateway )" != "ok" ]```.
This will be solved in next release.

rstanislav, 2018-10-31:
I have checked with latest commits - its works now, but something has happened(dont know if last commits changed something, or it was broken before) with ss-rules - shadowsocks not working, rules table always empty so all traffic is going thru omrvpn tunnel.

Ysurac, 2018-10-31:
True, a little that will be fixed before release :)

rstanislav, 2018-11-01:
Thanks a lot, now it works perfectly!

closed 2018-11-01 by rstanislav