Ysurac/openmptcprouter #203: Update to 0.40 crashed Router

Maudissure, 2018-11-01:
## Expected Behavior

Update to OMR V0.40

## Actual Behavior

After update, no access to OMR, same after new install on fresh SDHC.

## Steps to Reproduce the Problem

  1.Update to OMR 0.40
  2.No DHCP, no network connection (even with manual IP)

## Specifications

  - OpenMPTCProuter version: 0.40
  - OpenMPTCProuter VPS version: 0.61
  - OpenMPTCProuter platform: RPI3B+

Maudissure, 2018-11-01:
After another fresh install, i retrieved network access to OMR, i close.

Maudissure, 2018-11-01:
Nope ! DHCP not working. When i replace load balancing router with OMR router, nothing works. No DHCP, no interface connection.

SoGooDFR, 2018-11-01:
In my case, no problem with update 0.40 on x86_64 platform.

Ysurac, 2018-11-01:
Update from which version ?
Can you ping 192.168.100.1 ? If you change the IP, can you ping it ? Without DHCP, it's working or not ?

TheCrach, 2018-11-02:
/etc/init.d/network start must be done manually for some reason, I had the same problem (update from 0.34.1)

Ysurac, 2018-11-02:
I would need system log at least, and the way used to update.

Ysurac, 2018-11-02:
With a new install, all is working before you change the IP of OpenMPTCProuter (then you have no access to OpenMPTCProuter) ?
I can't reproduce the issue, can you give me the result of  ```uci show```using ssh on OpenMPTCProuter before changing the IP (or save the configuration and send me the file without server IP) ?

Maudissure, 2018-11-02:
My home subnet network is 192.168.1.* so i change immediatly the OMR subnet for testing. But after install i can access interface on 192.168.100.* with one client on this subnet.

As i said after installing OMR V0.40.1 AND VPS V.062, all is ok (except for Apple devices and company computers)

Maudissure, 2018-11-02:
A question : Can i force a device to use directly a WAN without using OMR ?

Ysurac, 2018-11-02:
With OMR not for now.

Maudissure, 2018-11-02:
Should be added to your todo list.

Ysurac, 2018-11-02:
I'm adding it, this will be in next release :)

Maudissure, 2018-11-02:
thanks a lot.

closed 2018-11-02 by Ysurac