Ysurac/openmptcprouter #205: DHCPDECLINE - Network Printer too !

Maudissure, 2018-11-03:
## Expected Behavior

 Using DHCP with network printer

## Actual Behavior

DHCP not working, even with DHCP static address (same behavior than Apple devices)

## Steps to Reproduce the Problem

The printer as IPV4 only activated.

Sat Nov  3 09:53:42 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sat Nov  3 09:53:54 2018 daemon.info dnsmasq-dhcp[1201]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 09:53:54 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 09:55:00 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 192.168.1.178 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:00 2018 daemon.info dnsmasq-dhcp[1201]: DHCPNAK(eth0) 192.168.1.178 00:1b:a9:e6:28:fc static lease available
Sat Nov  3 09:55:00 2018 daemon.info dnsmasq-dhcp[1201]: DHCPDISCOVER(eth0) 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:00 2018 daemon.info dnsmasq-dhcp[1201]: DHCPOFFER(eth0) 192.168.1.101 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:00 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 192.168.1.101 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:00 2018 daemon.info dnsmasq-dhcp[1201]: DHCPACK(eth0) 192.168.1.101 00:1b:a9:e6:28:fc Imprimante
Sat Nov  3 09:55:01 2018 daemon.info dnsmasq-dhcp[1201]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 09:55:01 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 09:55:05 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 192.168.1.101 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:05 2018 daemon.info dnsmasq-dhcp[1201]: DHCPACK(eth0) 192.168.1.101 00:1b:a9:e6:28:fc Imprimante
Sat Nov  3 09:55:06 2018 daemon.info dnsmasq-dhcp[1201]: DHCPDECLINE(eth0) 192.168.1.101 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:06 2018 daemon.warn dnsmasq-dhcp[1201]: disabling DHCP static address 192.168.1.101 for 10m
Sat Nov  3 09:55:15 2018 daemon.warn dnsmasq-dhcp[1201]: not using configured address 192.168.1.101 because it was previously declined
Sat Nov  3 09:55:19 2018 daemon.info dnsmasq-dhcp[1201]: DHCPDISCOVER(eth0) 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:19 2018 daemon.info dnsmasq-dhcp[1201]: DHCPOFFER(eth0) 192.168.1.178 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:19 2018 daemon.warn dnsmasq-dhcp[1201]: not using configured address 192.168.1.101 because it was previously declined
Sat Nov  3 09:55:19 2018 daemon.info dnsmasq-dhcp[1201]: DHCPDISCOVER(eth0) 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:19 2018 daemon.info dnsmasq-dhcp[1201]: DHCPOFFER(eth0) 192.168.1.178 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:19 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 192.168.1.178 00:1b:a9:e6:28:fc
Sat Nov  3 09:55:19 2018 daemon.info dnsmasq-dhcp[1201]: DHCPACK(eth0) 192.168.1.178 00:1b:a9:e6:28:fc Imprimante
Sat Nov  3 09:55:19 2018 daemon.warn dnsmasq-dhcp[1201]: not giving name Imprimante.lan to the DHCP lease of 192.168.1.178 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.101
Sat Nov  3 09:55:20 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 192.168.1.103 d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:20 2018 daemon.info dnsmasq-dhcp[1201]: DHCPNAK(eth0) 192.168.1.103 d4:3d:7e:be:0e:d2 address not available
Sat Nov  3 09:55:21 2018 daemon.info dnsmasq-dhcp[1201]: DHCPSOLICIT(eth0) 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:21 2018 daemon.info dnsmasq-dhcp[1201]: DHCPADVERTISE(eth0) fda5:1d90:439b::6703 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPDISCOVER(eth0) d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPOFFER(eth0) 192.168.1.229 d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPDISCOVER(eth0) d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPOFFER(eth0) 192.168.1.229 d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPSOLICIT(eth0) 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPADVERTISE(eth0) fda5:1d90:439b::6703 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 192.168.1.229 d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPACK(eth0) 192.168.1.229 d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.warn dnsmasq-dhcp[1201]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 09:55:25 2018 daemon.warn dnsmasq-dhcp[1201]: not giving name Imprimante.lan to the DHCP lease of 192.168.1.178 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.101
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPSOLICIT(eth0) 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPADVERTISE(eth0) fda5:1d90:439b::6703 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPSOLICIT(eth0) 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPADVERTISE(eth0) fda5:1d90:439b::6703 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPSOLICIT(eth0) 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPADVERTISE(eth0) fda5:1d90:439b::6703 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREQUEST(eth0) 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2
Sat Nov  3 09:55:25 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREPLY(eth0) fda5:1d90:439b::6703 00:01:00:01:1c:6d:6b:70:d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 09:55:25 2018 daemon.warn dnsmasq-dhcp[1201]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 09:55:25 2018 daemon.warn dnsmasq-dhcp[1201]: not giving name Imprimante.lan to the DHCP lease of 192.168.1.178 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.101
Sat Nov  3 09:55:54 2018 daemon.err /usr/bin/ss-redir[4339]: remote recv: Connection reset by peer
Sat Nov  3 09:55:57 2018 daemon.err /usr/bin/ss-redir[4341]: remote recv: Connection reset by peer
Sat Nov  3 09:55:58 2018 daemon.err /usr/bin/ss-redir[4341]: remote recv: Connection reset by peer
Sat Nov  3 09:56:02 2018 daemon.err /usr/bin/ss-redir[4341]: remote recv: Connection reset by peer
Sat Nov  3 09:56:03 2018 daemon.err /usr/bin/ss-redir[4340]: remote recv: Connection reset by peer
Sat Nov  3 09:56:03 2018 daemon.err /usr/bin/ss-redir[4339]: remote recv: Connection reset by peer
Sat Nov  3 09:56:11 2018 daemon.info dnsmasq-dhcp[1201]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 09:56:11 2018 daemon.info dnsmasq-dhcp[1201]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16

## Specifications

  - OpenMPTCProuter version: 0.40.1
  - OpenMPTCProuter VPS version: 0.62
  - OpenMPTCProuter platform: RPI3B+

Ysurac, 2018-11-03:
I would need the result of ```uci show network``` and ```uci show dhcp```.
You should also check if you have no other DHCP server on the network, the printer decline the ip address offered.

Maudissure, 2018-11-03:
No other DHCP server on network.

Maudissure, 2018-11-03:
After hard reboot of OMR , works again : 

Sat Nov  3 10:11:28 2018 daemon.info dnsmasq-dhcp[4819]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:11:28 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:11:40 2018 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sat Nov  3 10:11:54 2018 user.notice post-tracking-post-tracking: Set server 51.254.127.83 route via wan1 metric 1
Sat Nov  3 10:12:03 2018 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sat Nov  3 10:12:03 2018 user.notice post-tracking-post-tracking: Get status and settings for wan3... Done
Sat Nov  3 10:12:29 2018 daemon.err uhttpd[2250]: luci: accepted login on /admin/status/syslog for root from 192.168.1.100
Sat Nov  3 10:12:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:12:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:13:41 2018 daemon.info dnsmasq-dhcp[4819]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:13:41 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:13:59 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREQUEST(eth0) 192.168.1.229 d4:3d:7e:be:0e:d2
Sat Nov  3 10:13:59 2018 daemon.info dnsmasq-dhcp[4819]: DHCPACK(eth0) 192.168.1.229 d4:3d:7e:be:0e:d2 PC_Bureau
Sat Nov  3 10:13:59 2018 daemon.warn dnsmasq-dhcp[4819]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 10:14:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREQUEST(eth0) 192.168.1.178 00:1b:a9:e6:28:fc
Sat Nov  3 10:14:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPNAK(eth0) 192.168.1.178 00:1b:a9:e6:28:fc static lease available
Sat Nov  3 10:14:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPDISCOVER(eth0) 00:1b:a9:e6:28:fc
Sat Nov  3 10:14:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPOFFER(eth0) 192.168.1.101 00:1b:a9:e6:28:fc
Sat Nov  3 10:14:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREQUEST(eth0) 192.168.1.101 00:1b:a9:e6:28:fc
Sat Nov  3 10:14:32 2018 daemon.info dnsmasq-dhcp[4819]: DHCPACK(eth0) 192.168.1.101 00:1b:a9:e6:28:fc Imprimante
Sat Nov  3 10:14:32 2018 daemon.warn dnsmasq-dhcp[4819]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 10:14:49 2018 daemon.info dnsmasq-dhcp[4819]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:14:49 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:15:57 2018 daemon.info dnsmasq-dhcp[4819]: DHCPCONFIRM(eth0) 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16
Sat Nov  3 10:15:57 2018 daemon.info dnsmasq-dhcp[4819]: DHCPREPLY(eth0) fda5:1d90:439b::bb1d 00:01:00:01:22:d1:db:df:d0:d2:b0:83:83:16

Ysurac, 2018-11-03:
ok, so no idea why printer decline the ip address...

Maudissure, 2018-11-03:
Some strange behavior with a DHCP call from my iphone : 

Sat Nov  3 10:36:45 2018 daemon.info dnsmasq-dhcp[28902]: DHCPACK(eth0) 192.168.1.142 b4:8b:19:1b:4e:ee iPhoneauMaudit
Sat Nov  3 10:36:45 2018 daemon.warn dnsmasq-dhcp[28902]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 10:36:50 2018 daemon.info dnsmasq-dhcp[28902]: DHCPDECLINE(eth0) 192.168.1.142 b4:8b:19:1b:4e:ee
Sat Nov  3 10:36:50 2018 daemon.warn dnsmasq-dhcp[28902]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 10:37:03 2018 daemon.info dnsmasq-dhcp[28902]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Sat Nov  3 10:37:03 2018 daemon.info dnsmasq-dhcp[28902]: DHCPOFFER(eth0) 192.168.1.143 b4:8b:19:1b:4e:ee
Sat Nov  3 10:37:03 2018 daemon.info dnsmasq-dhcp[28902]: DHCPDISCOVER(eth0) b4:8b:19:1b:4e:ee
Sat Nov  3 10:37:03 2018 daemon.info dnsmasq-dhcp[28902]: DHCPOFFER(eth0) 192.168.1.143 b4:8b:19:1b:4e:ee
Sat Nov  3 10:37:04 2018 daemon.info dnsmasq-dhcp[28902]: DHCPREQUEST(eth0) 192.168.1.143 b4:8b:19:1b:4e:ee
Sat Nov  3 10:37:04 2018 daemon.info dnsmasq-dhcp[28902]: DHCPACK(eth0) 192.168.1.143 b4:8b:19:1b:4e:ee iPhoneauMaudit
Sat Nov  3 10:37:04 2018 daemon.warn dnsmasq-dhcp[28902]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103
Sat Nov  3 10:37:10 2018 daemon.info dnsmasq-dhcp[28902]: DHCPDECLINE(eth0) 192.168.1.143 b4:8b:19:1b:4e:ee
Sat Nov  3 10:37:10 2018 daemon.warn dnsmasq-dhcp[28902]: not giving name PC_Bureau.lan to the DHCP lease of 192.168.1.229 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.1.103

Why OMR try to give PC_Bureau name to the iphone ?

Ysurac, 2018-11-03:
Something with IP 192.168.1.229 is asking for the name PC_Bureau.lan but the name is already taken by something at 192.168.1.103.
It's not the same with the iPhone that DHCP server give 192.168.1.142, but the iPhone decline it, so DHCP server offer 192.168.1.143 but iphone decline it too.

Ysurac, 2018-11-03:
Something else at ip 192.168.1.229 seems to request this name.