Ysurac/openmptcprouter #206: Kernel strange error + drop connection every refresh

SoGooDFR, 2018-11-03:
## Actual Behavior

This is what happen during the DNS test on: https://ipleak.net/
After page refresh, the WAN2 (4G) drop.

> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Network device 'tun0' link is down
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: omrvpn (3048): udhcpc: received SIGTERM
> Sat Nov  3 17:41:03 2018 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Network device 'tun0' link is up
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'omrvpn' has link connectivity
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'omrvpn' is setting up now
> Sat Nov  3 17:41:03 2018 daemon.err glorytun-udp[5851]: running on device tun0 as pid 5851
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: omrvpn (5874): udhcpc: started, v1.28.3
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: omrvpn (5874): udhcpc: sending discover
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' is now down
> Sat Nov  3 17:41:03 2018 kern.err kernel: [ 5273.171258] mptcp_init4_subsockets: token 0x3871a6b5 bind() to 192.168.8.100 index 39 failed, error -99
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' is disabled
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' has link connectivity loss
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' has link connectivity
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' is enabled
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' is setting up now
> Sat Nov  3 17:41:03 2018 daemon.notice netifd: Interface 'wan2' is now up

## Steps to Reproduce the Problem

  1. Fresh install
  2. Configure OMR-Bypass + SQM
  3. Open https://ipleak.net/.

## Specifications

  - OpenMPTCProuter version: 40.2
  - OpenMPTCProuter VPS version: 0.61
  - OpenMPTCProuter platform: x86_64

Ysurac, 2018-11-03:
Can't reproduce the issue. What do you have in kernel log ?
The site is bypassed ?
Glorytun UDP is up at the same time, OMR is configured since a few second ?

SoGooDFR, 2018-11-03:
Kernel:
> Sat Nov 3 17:41:03 2018 kern.err kernel: [ 5273.171258] mptcp_init4_subsockets: token 0x3871a6b5 bind() to 192.168.8.100 index 39 failed, error -99

Only netflix is bypassed. When i start the dns test this log lines appear:
> Sat Nov 3 17:41:03 2018 daemon.notice netifd: Network device 'tun0' link is down
> Sat Nov 3 17:41:03 2018 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
> Sat Nov 3 17:41:03 2018 [...]

Ysurac, 2018-11-03:
This kernel error is not a problem : the interface is down, MPTCP can't use it. There is nothing more in kernel log ?

SoGooDFR, 2018-11-03:
Nothing more.

Ysurac, 2018-11-03:
So no idea. The interface is detected to be down by the kernel, check cable, switch,... if you use real interface.

SoGooDFR, 2018-11-03:
I change internal ip to my 4G router and no problem at all.

Ysurac, 2018-11-03:
Doesn't seems to be a software problem. 
If you have more logs, info or tests please reopen the issue.

closed 2018-11-03 by Ysurac

SoGooDFR, 2018-11-03:
Problem found ! My poor ADSL make this.