Ysurac/openmptcprouter #211: Add quota support

PxxBxx, 2018-11-08:
With 4G Wans having monthly quota, it would be great to set lower bandwidth when approaching quota, and disable Wan when quota is overrun.
Of course, it has to be a date-based quota, with a flexible period (1month most of the time). 
There is the issue of rebooting the router that reset the counters also.. Some persistancy have to be set.
It's a nice to have issue, but it would be great (for me and my 3x 4G Wans ^^)

PxxBxx, 2018-11-08:
Best RTFM ever, thanks for being educative. 
I'll go and look into omr-quota to see if this feature is what I need, I'll let you know if I have any issues. 
Damn... Really sorry!

MrDaweb, 2018-11-10:
Hello. What it would miss is the reset date. Often it is not the first of the month. How perfect it would be :)

Ysurac, 2018-11-10:
You have really case when reset date is not first of month ? This will be a little be more difficult...
I don't like quota ;)

MrDaweb, 2018-11-10:
Je répond en français je suis meilleur :)
Mais oui j'ai deux abonnements 4G que je souhaite rajouter à mon vdsl pour tester, et les deux expirent chacun à une date différente !

Ysurac, 2018-11-10:
Il y a des fournisseurs avec des quotas mensuels qui sont pas au 1er du mois et que l'on peut pas bloquer en cas de dépassement ?

MrDaweb, 2018-11-10:
En fait si on prend des forfaits bandyou ou SFR Red par exemple, ça passe en débit fortement réduit une fois dépassé le quota, mais du coup il faudrait pouvoir re-regler à ce moment là le debit max sinon ca risque de mal fonctionner je pense. Alors que du coup l'utilisation quota bloque simplement. Mais hélas, sauf à réussir à prendre démarrer l'abonnement pour le 1er du mois (pas simple !) ca renouvelle à la date de démarrage. Par exemple : "Votre forfait sera remis à zéro le 23/11"

MrDaweb, 2018-11-11:
Merci ;)
Par ce que même si c’est pas un gros soucis, on ne peut pas automatiser du coup.

closed 2018-12-26 by Ysurac