Ysurac/openmptcprouter #217: odhcpd multiple lan settings

PxxBxx, 2018-11-14:
> I ***know*** I ask others to do things I could do myself, but I'm a limited "ubuntu/debian" user, not an expert in embedded linux....

I would love to have DHCP leases outside the "lan" network.
eg.
- set-top boxes that are bound to an ISP router (a specific wan network).
- surveillance products that are also bound to an ISP router (because they sell the package, it's a black box for us, users... no setup possible)
- of course, all this because I want this modem router as a wan for OMR, but also need to benefit from it outside of OMR, and keep only one home LAN.

Problem : OMR requires that dhcp are disabled on all wans.

So the goal is : OMR should be able to assign IPs outside the OMR network.

I only dig at this problem on a theorically point of view, maybe the technical reality will invalidate what follows.

> because I know I already asked the same questions one month ago..... but I thought a lot about it since.

- a dhcp server cannot assign IPs (static or not) outside its network ("lan" in odhcpd setup)
- But a dhcp server can handle multiple networks in its configuration (I managed that with udhcpcd, dhcpd, ... on some Linux distros)

Would it be possible for the luci dhcp configuration page, to allow the specification of out-of-bound network IP leases ?

eg : 
1. OMR has its nominal 192.168.100.0/24 network
2. I setup one or mote static leases (via MAC address) with IPs in the 192.168.3.0/24 subnet, with the ISP router as its gateway (eg. IP=192.168.3.2 GW=192.168.3.1 NM=255.255.255.0)
3. OMR detects a new 192.168.3.0/24 network
4. OMR add a new network config, and setup whatever is needed in odhcpd, with no IP pool reserved, only the static leases for this subnet (so no dhcp request will be answered for this subnet, except from the devices previously declared)
5. these declared devices (via their MAC address) will obtain static IP leases, and will talk directly to their gateway, the ISP router. 
6. If I remove all the 192.168.3.0/24 static lease, the network is removed from odhcpd

Dear @Ysurac I know you will hate me for asking this a second time, but If you still say "no", I promise I won't ask again (or I will create a new GH account ^^). I only sent this request because I'm confident this should work. But it could also be overkill, and you know better than me.

Ysurac, 2018-11-14:
OpenMPTCProuter doesn't require DHCP disabled on all wans, but if all wans are on same network you will have a problem. Static DHCP is not a problem.

It's dnsmasq that is used as DHCP server.

You can add new interface (in *Network->Interfaces*) that you can add in lan firewall zone and use IP in the network you want and set DHCP server on it as you want.

So I don't really see what problem you can't solve with current OpenMPTCProuter here :)

PxxBxx, 2018-11-14:
sorry for mentioning odhcpd instead of dnsmasq, i'm not familiar with OMR linux base, I did a simple "ps" and saw a odhcpd process...
I think you gave me enough indications to manage it by myself.
What I asked was some "magic UI" that would detect I setup new subnet in the dhcp leases, and do what you list : add a new interface, setup the firewall zones, setup dhcp server on this subnet.

seems simple enough to do it myself !
> note to myself : to backup the OMR config or the sdcard before clicking everywhere....

closed 2019-02-20 by Ysurac