Ysurac/openmptcprouter #218: Status interface down to email not work

ThibautBascoulergue, 2018-11-15:
## Expected Behavior

  In English please.

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Ysurac, 2018-11-15:
Really ? an empty bug reports with no details ?

ThibautBascoulergue, 2018-11-15:
lol sorry failed past :)

ThibautBascoulergue, 2018-11-15:
nothing in log after restart or stop and start

Ysurac, 2018-11-15:
What is the result of `uci show mail` ? (without secret info)

ThibautBascoulergue, 2018-11-15:
[image]

Ysurac, 2018-11-15:
All seems good, you should have a `/etc/msmtprc`, at least when you run `/etc/init.d/msmtp restart`

ThibautBascoulergue, 2018-11-15:
mmh :/ another problem :+1: 
[image]

Ysurac, 2018-11-15:
I think this time the problem is not on my side. Check your settings and if your VPS provider doesn't block some ports.

ThibautBascoulergue, 2018-11-15:
ok thx

Ysurac, 2018-11-15:
OpenMPTCProuter v0.41.1 will fix msmtprc creation.
I will try with a gmail account if it's working (on another provider than scaleway that blocks all smtp ports).

ThibautBascoulergue, 2018-11-15:
ok fine

closed 2019-02-20 by Ysurac