Ysurac/openmptcprouter #223: Impossible de me connecter au VPS OVH depuis mon réseau interne sur OMR

mykelinho, 2018-11-23:
OpenMPTCProuter v0.41 2018-11-12 18:19
OpenMPTCProuter VPS v0.64:
RPI3+
VPS OVH 

Bonjour,
Je ne peux plus me connecter à mon VPS depuis mon réseau WIFI/LAN lié au routeur OPENMTCPROUTER.
Sur putty, WinSCP... La connexion reste refusé.
Je peux me connecté uniquement sur un réseau externe à OMR (4G, ADSL...)
Est-ce un mauvais réglage dans OMR? Je n'ai pourtant touché à rien, mis à part avoir fait la dernière mise à jour.

mykelinho, 2018-11-24:
effectivement, mais sur putty je tente de me connecter en direct avec l'ip de mon vps et le 65222 ça ne fonctionne pas, alors que si je me connecte à un réseau externe (exemple 4G) ça fonctionne

Ysurac, 2018-11-24:
Et depuis OpenMPTCProuter `ssh -p 65222 tonip` ?

Ysurac, 2018-11-24:
Tu essayes par l'IPv4 ou le nom ? Et c'est bien la même IPv4 qui est renseigné dans le wizard ?

mykelinho, 2018-11-24:
j'essaye avec l'adresse IPv4 et c'est bien la même qui est inscrite dans le wizard

Ysurac, 2018-11-24:
Je sais pas la. 
L'IP du VPS étant "bypasser" partout (sinon shadowsocks et le VPN ne peuvent pas se connecter dessus), et l'IP fonctionnant depuis OpenMPTCProuter (donc les routes sont bonnes)...
Pas de configuration "étrange" niveau firewall ?
Que donne un `traceroute ipduvps` depuis le client ? Normalement ça doit passer directement par un des WANs.

mykelinho, 2018-11-24:
Niveau par feu j'ai ouvert le port 80 et 8080 sur 2 ips fixe de mes appareils, rien de plus
Voici le résultat du traceroute 
[image]

mykelinho, 2018-11-24:
et les configs du par feu 
[image]
[image]

[image]

Ysurac, 2018-11-24:
Le traceroute est pas bon : tu passes par le VPN.
Il me faudrait un `ip route`, normalement il y a une route vers l'ip de ton VPS avec une metric de 1, mais ça doit pas être le cas. Il faut que je trouve dans quel cas cette route n'est pas créé...

mykelinho, 2018-11-24:
merci, quelle est la commande iproute IP ?

Ysurac, 2018-11-25:
Sur OpenMPTCProuter par ssh, juste `ip route`. Ca donne toutes les routes :)

mykelinho, 2018-11-25:
voici le résultat ci dessous, 
il semblerait que les 2 VPS que je possèdent soit utilisé alors que j'ai remplacer le OVH 54.xxx par Firstheberg 86.xxx
[image]

mykelinho, 2018-11-25:
j'ai finalement réinstallé l'image openmptcrouter sur mon raspberry et j'ai resaisi mes paramètres (dhcp, firewall)... et ça refonctionne bien.
C'était probablement du à une fausse manip je pense.

closed 2018-11-25 by mykelinho

vinceducat, 2020-02-25:
j ai le meme probleme a par une reinstallation comment faire