Ysurac/openmptcprouter #226: TCP connection was reset by openwrt

IOTBaicheng, 2018-11-27:
## Expected Behavior

 I hope to  access web site (wget https://multipath-tcp.org) successfully when MPTCP is enabled.
  In English please.

## Actual Behavior

the tcp connection is reset by openwrt whith RST. 
But if I disable MPTCP  or set the path-manager to default or set the second WAN interface to handover or backup , I can access the web, but I can not see MPTCP option in TCP options.

And if I perform wget 
## Steps to Reproduce the Problem
  1. setup a MPTCP enviroment in my virtralbox. the setup is:
utuntu64 -----(lan)------ openwrt -----(2 wan interfaces)----- internet 
  2. MPTCP settings:
network.globals=globals
network.globals.ula_prefix='fdc6:8291:b14f::/48'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_checksum='1'
network.globals.mptcp_path_manager='fullmesh'
network.globals.multipath='enable'

network.wan1=interface
network.wan1.ifname='eth1'
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.macaddr='08:00:27:07:cd:26'
network.wan1.ipaddr='192.168.1.201'
network.wan1.gateway='192.168.1.1'
network.wan1.netmask='255.255.255.0'
network.wan1.multipath='master'

network.wan2=interface
network.wan2.ifname='eth2'
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.macaddr='08:00:27:ac:4e:7f'
network.wan2.ipaddr='192.168.1.202'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.1.1'
network.wan2.multipath='on'

the way to access web:
wget https://multipath-tcp.org

## Specifications
   (r7390-dd6d554b2c)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.41.1

  - OpenMPTCProuter version: v0.41.1
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2018-11-27:
multipath-tcp.org website is used to test if multipath work on each interface on some case (if tracebox check to the VPS doesn't work) by OMR status page. So this check can create a RST on connection.

IOTBaicheng, 2018-11-27:
Thank you for reply.
If I enabled MPTCP, I can not access other webs that don't support MPTCP, also reset by openwrt.
and I can't see any SYN packet goes to WAN interface.
so what is your suggestion to fix my problem.

Ysurac, 2018-11-27:
When I look your configuration, there is some problems: 
* wan1 and wan2 in same network
* wan1 and wan2 use same gateway

IOTBaicheng, 2018-11-27:
this is not allowed?
when I perform wget in openwrt, the MPTCP works well. both wan1 and wan2 can setup a MPTCP  flow.

Ysurac, 2018-11-27:
It's at least a strange configuration and this can make some strange results too. This would be far better to have each wan on his own network.
What results do you have in OpenMPTCProuter status page ?

IOTBaicheng, 2018-11-27:
In OpenMPTCProuter status, I haven't found any item that looks wrong.
which item do you want to check?

Ysurac, 2018-11-27:
Anything that doesn't give a green mark :)
But with your configuration it's possible that no error are detected.
With your current configuration I don't know how wan in same network can give a really working result, I never tested this case and make check to add errors about that.

IOTBaicheng, 2018-11-27:
do you ask for status in "http://192.168.56.4/cgi-bin/luci/admin/system/openmptcprouter/status"?
it shows wan1 and wan2 has the same gateway(192.168.1.1) and wan ip addr, the color is not green,. it has a picture of "!". and for wan1 and wan2 says no server ping after 1 second. and for mng interface of 192.168.56. 1 says your IP was not leased by openmptcprouter.

Ysurac, 2018-11-27:
True it's this page.
So the status page detect a problem with wans interfaces (the "not leased..." is not a problem).
Put a wan in another network with it's own gateway in the same network and this should solve the issue.

IOTBaicheng, 2018-11-27:
thank you very much. I will try and update here.

IOTBaicheng, 2018-12-01:
Hi Ysurac,
I recreate a new virtual machine, and use almost the same interface configuration, it will not get RST from opwnwrt now. Maybe some other mistakes caused this problem. I will close this issue.
thank you!

closed 2018-12-01 by IOTBaicheng