Ysurac/openmptcprouter #227: Which types of WAN interfaces are supported out of the box on RPI image ?

renaudguerin, 2018-11-28:
Salut :)

Congrats for the project, looks amazing !
I plan to use OMR on an RPI to share two 4G connections.

- I already have one 4G router with Ethernet ports, so that one will be easy.
- For the second 4G connection, I need to buy new hardware for the SIM and I'd like to know what the OMR image supports out of the box (or can easily be added) : ideally I'd like to use a cheap USB Huawei 4G dongle directly plugged into the RPI, or even hotplug an Android/iPhone in hotspot mode via USB or Bluetooth to quickly add more juice to my connection when I'm home ! I know that OverTheBox supports both dongle and phone hotspot over USB, is it the same for OMR ? If not, I'll buy a "real" home 4G router with Ethernet, or a Wifi one, but that's more expensive / complex. 

Thank you

Ysurac, 2018-11-28:
OpenMPTCProuter support all dongles/android/iPhone supported by OpenWrt by default.

But using an USB dongle on the RPI is really not a good idea. An ethernet 4G router is really better. Avoid wifi too, using 4G unstable connection with a wifi unstable connection can't really be stable :)

Ysurac, 2018-11-28:
USB on RPI is shared with ethernet port, so it's lot of data for USB. RPI need also to have a good power supply to give enough power to the dongle.
Same for phone hotspot via USB. It's working for some users but I never tested that.

closed 2018-12-13 by Ysurac