Ysurac/openmptcprouter #229: VPS config error 500

PxxBxx, 2018-11-29:
## Expected Behavior

0.42.1 new functionality to get settings/keys/status/... working

## Actual Behavior

I got a 500 when calling the vps admin script from the router

## Steps to Reproduce the Problem

  1. set the VPS key in the OMR interface
  2. check the system log
  3. confirm with `/etc/init.d/openmptcprouter-vps start` on the router

## Specifications

  - OpenMPTCProuter version: 0.42.1
  - OpenMPTCProuter VPS version: 0.66
  - OpenMPTCProuter platform: x86_64

## Traces/Logs

### VPS syslog
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]: 92.88.57.188 - - [29/Nov/2018 20:45:42] "POST /login HTTP/1.1" 200 -`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]: 92.88.57.188 - - [29/Nov/2018 20:45:42] "GET /config HTTP/1.1" 500 -`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]: Error on request:`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]: Traceback (most recent call last):`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 205, in run_wsgi`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:     execute(self.server.app)`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 193, in execute`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:     application_iter = app(environ, start_response)`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1997, in __call__`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:     return self.wsgi_app(environ, start_response)`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1985, in wsgi_app`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:     response = self.handle_exception(e)`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1540, in handle_exception`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:     reraise(exc_type, exc_value, tb)`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/_compat.py", line 33, in reraise`
`Nov 29 20:45:42 ardcode-omr omr-admin.py[591]:     raise value`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1982, in wsgi_app`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     response = self.full_dispatch_request()`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1614, in full_dispatch_request`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     rv = self.handle_user_exception(e)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1517, in handle_user_exception`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     reraise(exc_type, exc_value, tb)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/_compat.py", line 33, in reraise`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     raise value`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1612, in full_dispatch_request`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     rv = self.dispatch_request()`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1598, in dispatch_request`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     return self.view_functions[rule.endpoint](**req.view_args)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/local/lib/python3.5/dist-packages/flask_jwt_simple/view_decorators.py", line 25, in wrapper`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     return fn(*args, **kwargs)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/local/bin/omr-admin.py", line 97, in config`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     data = json.load(f)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3.5/json/__init__.py", line 268, in load`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3.5/json/__init__.py", line 319, in loads`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     return _default_decoder.decode(s)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3.5/json/decoder.py", line 339, in decode`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     obj, end = self.raw_decode(s, idx=_w(s, 0).end())`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3.5/json/decoder.py", line 357, in raw_decode`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     raise JSONDecodeError("Expecting value", s, err.value) from None`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]: json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]: 92.88.57.188 - - [29/Nov/2018 20:45:43] "POST /shadowsocks HTTP/1.1" 500 -`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]: Error on request:`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]: Traceback (most recent call last):`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 205, in run_wsgi`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:     execute(self.server.app)`
`Nov 29 20:45:43 ardcode-omr omr-admin.py[591]:   File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 193, in execute`
[... 71 lines cut ...]
`Nov 29 20:45:45 ardcode-omr shorewall[32648]: Setting up Accept Source Routing...`
`Nov 29 20:45:45 ardcode-omr shorewall[32648]: Preparing iptables-restore input...`
`Nov 29 20:45:45 ardcode-omr shorewall[32648]: Running /sbin/iptables-restore ...`
`Nov 29 20:45:45 ardcode-omr shorewall[32648]: IPv4 Forwarding Enabled`
`Nov 29 20:45:45 ardcode-omr root: Shorewall reloaded`
`Nov 29 20:45:45 ardcode-omr shorewall[32648]: done.`
`Nov 29 20:45:45 ardcode-omr systemd[1]: Reloaded Shorewall IPv4 firewall.`
`Nov 29 20:45:45 ardcode-omr omr-admin.py[591]: 92.88.57.188 - - [29/Nov/2018 20:45:45] "POST /shorewall HTTP/1.1" 200 -`
`Nov 29 20:45:45 ardcode-omr omr-admin.py[591]: 92.88.57.188 - - [29/Nov/2018 20:45:45] "POST /mptcp HTTP/1.1" 200 -`

### System log
Thu Nov 29 20:31:36 2018 user.notice mptcp: Reloading mptcp config due to ifup of omr6in4 (6in4-omr6in4)
Thu Nov 29 20:31:36 2018 user.notice unbound: default root hints (built in rootservers.net)
Thu Nov 29 20:31:36 2018 user.notice unbound: default memory configuration
Thu Nov 29 20:31:36 2018 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN"> <title>500 Internal Server Error</title> <h1>Internal Server Error</h1> <p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
Thu Nov 29 20:31:36 2018 daemon.notice unbound: [5884:0] notice: init module 0: iterator
Thu Nov 29 20:31:36 2018 daemon.info unbound: [5884:0] info: start of service (unbound 1.8.1).
Thu Nov 29 20:31:38 2018 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: { "reason": "changes applied", "result": "done" }
Thu Nov 29 20:31:38 2018 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: { "reason": "Invalid parameters", "result": "error" }

### Manual script execution
root@OpenMPTCProuter:~# /etc/init.d/openmptcprouter-vps start
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN"> <title>500 Internal Server Error</title> <h1>Internal Server Error</h1> <p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
{ "reason": "changes applied", "result": "done" }
{ "reason": "Invalid parameters", "result": "error" }

PxxBxx, 2018-11-29:
sorry, I know only perl/C/shell/php/js, i'm ashamed I can't understand/debug errors via python stack trace :-/
I won't be able to help, all I can do is report you the error....

Ysurac, 2018-11-29:
I think the error was caused by a trailing comma at the end of `/etc/shadowsocks-libev/config.json` that appear in some case. This should be fixed in latest OpenMPTCProuter VPS update.

closed 2018-11-29 by Ysurac