Ysurac/openmptcprouter #233: OMR doesn't working anymore

Maudissure, 2018-11-30:
## Expected Behavior

 Functionnal OMR

## Actual Behavior

On is own lan, nothing work, neither DHCP nor Internet connection

## Steps to Reproduce the Problem

  1.Fresh install and configuration on my old LAN (OMR IP : 192.168.1.2 - Old Router IP 192.168.1.1) - OMR interface can be accessed and configured.
  2.Unplug old router and reboot OMR box
  3.No DHCP
  4.With fixed IP on Windows Computer : No access to OMR interface, No internet connection

## Specifications

  - OpenMPTCProuter version: 0.42.5
  - OpenMPTCProuter VPS version: 0.65
  - OpenMPTCProuter platform: RPI3B+

Maudissure, 2018-12-01:
By the way i succed to connect some devices with OMR, but my wife's company computer still refuse to connect to internet (yes even with the MAC address on OMR Bypass to one Wan) and Apple devices don't work with OMR, plus the OMR Bypass doesn't work anymore with Netflix.

Considering all my devices works with my load balancing router , the OMR system is clearly involved. 

After three hours, trying to make it work without success, testing multiple configurations, i'm done. And i'm very sorry, this system seemed to me the ultimate solution to my poor ADSL connection. 

I'll checked out only when the 1.0 version will be out. And if it don't work in my LAN environment, too bad for me.

Thanks for the work, i know how this kind of project can be difficult.

closed 2018-12-01 by Maudissure

Ysurac, 2018-12-01:
Without help on debug this will never be solved. I can't have all existing configuration.

Maudissure, 2018-12-02:
I completely understand that you can't have all existing configurations, my problem is purely personnal. My home is entirely connected to the LAN with multiple devices. Beside this, me and my wife works at home. So i can't leave a central device like OMR in alpha state 24/24 connected.

Yesterday my wife's company VPN just revoked is credentials and i don't know why. This appened with OMR as home router during testing. So i just can't keep OMR for now on my LAN; too much risks.

If  you want, i'll could make ponctual testing for you, but no more. So don't hesitate to contact me if needed. (mlemaudit1@gmail.com , in french please :D)