Ysurac/openmptcprouter #234: Fix Netflix over OMR-Bypass

Ysurac, 2018-12-01:
I can test for one month and try to fix it.
Answer to ping without problem, https not working.

Ysurac, 2018-12-01:
Bug found ! It's a problem in iptables rules. Some rules seems to be added sometimes that make omr-bypass fail. Need to find what and when and will be fixed fast. A reboot fix problem for a little time...

Ysurac, 2018-12-01:
Maybe not in iptables. Work when glorytun is stopped or when default route is not the vpn.
So this seems more like a route problem.

Ysurac, 2018-12-02:
Working with a fresh configuration but some bugs found that can cause problems in some cases.

Ysurac, 2018-12-10:
Fixed in latest release.

closed 2018-12-10 by Ysurac

cron410, 2018-12-29:
I'm able to use netflix without any custom bypass rules added. Only what you provide by default.