Ysurac/openmptcprouter #236: Configuration IP TV  Box (FREE french operator) with OpenMPTCPRouter to TV stream

Vinc32du44, 2018-12-02:
## Expected Behavior

  IPTV Box (FREE French operator) access or bypass to FREE Modem Server

## Actual Behavior

I try to explain my problem. (Sorry for my english ...)
All my network (DHCP, aggregate with ADSL and 4G router, etc ...) works properly but my box tv on the network does not want to connect to the WAN1 (FREE modem) to recover the TV stream.
So I created a static lease with a redirection to the WAN1 but it does not work.
I set OMR-Bypass with IP TV BOX to WAN1, MAC addressed to WAN1 but still no TV stream.
Please help me: D
It seems to me that at FREE there is a history of VLAN100 so that the TV player can access the TV content of the modem but I can not put this in place.
After trying several tutorial I am at the end of solutions.
Do you have ideas to submit?

cordially
Vincent

## Steps to Reproduce the Problem

  1. Connect TV player to the switch
  2. Check if there an IP adress => OK
  3. No TV stream on player

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.42.7 r7405-06a20afb34
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS v0.65
  - OpenMPTCProuter platform: RPI3

Ysurac, 2018-12-02:
If you can set a static IP to the tv box (I don't remember if it's possible), you can set it on the same network as the freebox server. Else you can create an interface on OpenMPTCProuter and use DHCP to give a static IP to the freebox tv in the same network of the freebox server.
If the freebox are on there own same network this will work.

Vinc32du44, 2018-12-02:
Thanks for your answer :)
I can't set a static IP on TV box. My freebox server is on WAN1 (192.168.10.1) with DHCP off and TV box static IP (192.168.100.212). Not same network 192.168.10.x / 192.168.100.x
Can i set DHCP on (freebox server) and redirect TV box IP to WAN1.
Which interface should I create (name and parameter ...) for it to work. Is it possible have more detail ? I am a little lost !

Thanks a lot :D

Vinc32du44, 2018-12-02:
Ok TV box work !!! (Android TV)=> Thanks a lot ;)
But when i launch freebox TV =>No stream ???
IP TV Box 192.168.10.2 IP FREEBOX 192.168.10.1 (same network)
I thinks that TV Box does not see the tv stream...
It is very likely that comes into play the history of the VLAN 100 but how to put it up;)

Ysurac, 2018-12-02:
I don't have a Freebox with a player, only an old one alone :)
If your ethernet cable is on the freebox server, vlan should not be removed anywhere, else you may have to set it on the switch if it is manageable.
No idea, but not really related to OpenMPTCProuter. If I understand well your configuration, there is the Freebox server, a cable and the TV box now.

Vinc32du44, 2018-12-02:
Ok I found the solution! Redirect IP from the tv box to wan1 with OMR-bypass and everything is back in order. in fact it was just assigning the IP of the tv box on the same network as the Freebox server and redirecting the streams to the server by bypassing openmptcprouter that's the solution! I hope it can help others with this post. in any case thank you for your help and responsiveness. Good luck for your future progress

ccdu35, 2018-12-30:
Bonjour, 
Je m'excuse de rebondir sur ce post mais il soulève chez moi une question : Vincent, dans votre réseau le freebox player bénéficie t'il de l’agrégation ? Je ne parviens pas à savoir si vous voulez juste pouvoir continuer à accéder au service TV et/ou augmenter le débit allouer à ce service
Je viens en effet de migrer vers la freebox Delta qui remplace ma FB révolution. J'ai migré pensant que l’agrégation 4G+xdsl incluse me rendrait éligible à la TV hors TNT via freebox player. Ce n'est malheureusement pas le cas, la 4G free (que ma box ne capte pas à son emplacement ) ne bénéficie pas au service TV. Mon débit ADSL est de 2 Mb. 
J'ai par ailleurs un routeur HUAWEI 4G et une antenne 4G extérieure qui me délivre de 80 à 150 mb via le réseau free et j'utilise un chromecast pour diffuser les chaines hors TNT mais ce n'est pas l'idéal (pas d'enregistrement...)
Vu mon débit ADSL, l'agregation n'a pas d’intérêt pour moi  en revanche si cela me permet de bénéficier du service TV, replay et VOD avec un débit autour des 100mb je suis preneur.
Mes connaissances réseaux sont limitées, j'ai un raspberry pi3 B+ et avant de prendre un VPS je souhaite m'assurer de la faisabilité. 
A moins que ce soit possible avec un switch manageable comme j'ai pu le lire sur des forums. Mais là aussi je ne sais pas s'il s'agit juste de continuer à bénéficier du service ou en augmenter le débit.
Par avance merci de votre retour.
Cordialement.
CC

ccdu35, 2018-12-31:
Bonjour,
Merci pour votre retour. Cela peut donc être quand même intéressant pour stabiliser ma connexion.
Merci encore. 
Cordialement.
CC

Minas103, 2020-09-26:
Bonjour à tous,

Désolé d'arriver plusieurs mois après la bataille, mais je suis dans le même cas que ccdu35 :
 - freebox delta avec une ligne ADSL à 3mb mais un ping à 35/45ms, une réception 4G (delta) à 30mb mais limité à 200Go,
 - un router 4G avec une ligne "4G illimitée" avec un débit entre 60 et 115mb
 - VPS OVH

Le but étant de permettre à mon fils de jouer en ligne sans trop de latence, et de conserver du débit après ses grosses et innombrables mises à jour de jeux.
En branchant tout sur un switch et en coupant le DHCP de la box et du router, j'arrive à agréger les connexions (même si le débit ne me semble souvent en dessous du débit 4G router), mais le Player n'arrive pas se synchroniser sur la delta.
J'ai bien compris que  Vinc32du44 a résolu ce problème en redirigeant l'IP du player sur le wan1 (sa box) via OMR-bypass ... mais là, techniquement je coince (... malheureusement la nature ne nous a pas tous doté des mêmes talents).
Pouvez-vous m'expliciter un peu plus la démarche SVP.

Cordialement

Minas103, 2020-09-26:
Je pense avoir résolu le problème de synchronisation du player, (comme évoqué plus haut) avec OMR-Bypass
 \Services\OMR-Bypass\
           > Adresse MAC (du player) / Interface : wan 1 (Box delta)

JeedHome44, 2020-10-26:
Bonjour,

J'ai essayé de comprendre l'anglais au début mais pas tout compris.

J'ai tenté d'allumer mon Free player Mini 4k hier et le ventilo tourne à fond en continu. Je pense qu'il ne trouve pas la Freebox.

J'ai fixé l'ip du Freeplayer sur le DHCP OMR. J'ai ensuite fait un By-Pass avec le Mac du Freeplayer pour le diriger toujours vers le wan de Free.

J'ai compris qu'il ne fallait pas faire ça sur OMR mais sur la Freebox. Il faut donc que ce soit la Freebox qui donne un IP au Freeplayer. Mais sachant qu'il faut un seul DHCP actif sur le réseau, je ne comprends pas comment fixer l'ip du Freeplayer sur le LAN de la Freebox et non le LAN de OMR.
Ma Freebox est en 192.168.1.1

Il faudrait donc que je mette le Freeplayer en 192.168.1.2 par exemple mais ou?

Merci