Ysurac/openmptcprouter #237: Port forwarding !

Roken120, 2018-12-03:
Hello, port forwarding not work.
[image]
Please help and i have a error in log

Mon Dec  3 00:03:59 2018 daemon.notice openvpn(omr)[23560]: Exiting due to fatal error
Mon Dec  3 00:04:04 2018 daemon.notice openvpn(omr)[23583]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:04 2018 daemon.notice openvpn(omr)[23583]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:04 2018 daemon.warn openvpn(omr)[23583]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:04 2018 daemon.err openvpn(omr)[23583]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:04 2018 daemon.notice openvpn(omr)[23583]: Exiting due to fatal error
Mon Dec  3 00:04:09 2018 daemon.notice openvpn(omr)[23662]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:09 2018 daemon.notice openvpn(omr)[23662]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:09 2018 daemon.warn openvpn(omr)[23662]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:09 2018 daemon.err openvpn(omr)[23662]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:09 2018 daemon.notice openvpn(omr)[23662]: Exiting due to fatal error
Mon Dec  3 00:04:14 2018 daemon.notice openvpn(omr)[23712]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:14 2018 daemon.notice openvpn(omr)[23712]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:14 2018 daemon.warn openvpn(omr)[23712]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:14 2018 daemon.err openvpn(omr)[23712]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:14 2018 daemon.notice openvpn(omr)[23712]: Exiting due to fatal error
Mon Dec  3 00:04:19 2018 daemon.notice openvpn(omr)[23741]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:19 2018 daemon.notice openvpn(omr)[23741]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:19 2018 daemon.warn openvpn(omr)[23741]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:19 2018 daemon.err openvpn(omr)[23741]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:19 2018 daemon.notice openvpn(omr)[23741]: Exiting due to fatal error
Mon Dec  3 00:04:24 2018 daemon.notice openvpn(omr)[23802]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:24 2018 daemon.notice openvpn(omr)[23802]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:24 2018 daemon.warn openvpn(omr)[23802]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:24 2018 daemon.err openvpn(omr)[23802]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:24 2018 daemon.notice openvpn(omr)[23802]: Exiting due to fatal error
Mon Dec  3 00:04:29 2018 daemon.notice openvpn(omr)[23862]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:29 2018 daemon.notice openvpn(omr)[23862]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:29 2018 daemon.warn openvpn(omr)[23862]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:29 2018 daemon.err openvpn(omr)[23862]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:29 2018 daemon.notice openvpn(omr)[23862]: Exiting due to fatal error
Mon Dec  3 00:04:34 2018 daemon.notice openvpn(omr)[23955]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:34 2018 daemon.notice openvpn(omr)[23955]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:34 2018 daemon.warn openvpn(omr)[23955]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:34 2018 daemon.err openvpn(omr)[23955]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:34 2018 daemon.notice openvpn(omr)[23955]: Exiting due to fatal error
Mon Dec  3 00:04:39 2018 daemon.notice openvpn(omr)[23996]: OpenVPN 2.4.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Mon Dec  3 00:04:39 2018 daemon.notice openvpn(omr)[23996]: library versions: OpenSSL 1.0.2p  14 Aug 2018, LZO 2.10
Mon Dec  3 00:04:39 2018 daemon.warn openvpn(omr)[23996]: ******* WARNING *******: All encryption and authentication features disabled -- All data will be tunnelled as clear text and will not be protected against man-in-the-middle changes. PLEASE DO RECONSIDER THIS CONFIGURATION!
Mon Dec  3 00:04:39 2018 daemon.err openvpn(omr)[23996]: ERROR: Cannot ioctl TUNSETIFF tun0: Resource busy (errno=16)
Mon Dec  3 00:04:39 2018 daemon.notice openvpn(omr)[23996]: Exiting due to fatal error

Roken120, 2018-12-03:
The last version

kamcio2603, 2018-12-03:
Please update issue with info 
OpenMPTCProuter version: (Last is not a version)
OpenMPTCProuter VPS version: (Last is not a version)
OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Ysurac, 2018-12-03:
So it should be 0.42.7 and 0.72 for the VPS.
You seems to have changed default VPN to OpenVPN but there is no key set for OpenVPN.
Why do you want OpenVPN ? It's slower than Glorytun TCP. If you really really want OpenVPN, you have to add the OpenVPN option to VPS install script.

Roken120, 2018-12-03:
x86_64
No i use glorytun
I you send more information in two hours

Roken120, 2018-12-03:
The vps is in debian 9

Ysurac, 2018-12-03:
If glorytun is used OpenVPN is disabled by default, so you shouldn't have these logs.
You should check in OpenMPTCProuter wizard by checking *advanced settings* checkbox.
What is also the status of all in OpenMPTCProuter status page ?

Roken120, 2018-12-03:
[Image]

[Image]

Télécharger Outlook pour Android<https://aka.ms/ghei36>

Roken120, 2018-12-03:
https://s.amsu.ng/mWfVbWFeAFBN

Roken120, 2018-12-03:
The vos is from firstheberg

Ysurac, 2018-12-03:
Disable openvpn that seems to be enable: `uci set openvpn.omr.enabled=0`
Follow https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding to enable port forwarding and check `/etc/shorewall/rules`, there was an issue in previous release to comment/uncomment this files in VPS script (you may need to update it to latest version).

Roken120, 2018-12-03:
All work maybe the 4G sfr 
[image]

Ysurac, 2018-12-03:
This is not a real problem, it's only a check. The SFR connection seems to have a high latency, so test can fail.

Roken120, 2018-12-03:
For ipv6 ?
Vps hosted by firstheberg.

Roken120, 2018-12-03:
/etc/shorewall/params.vpn
VPS_ADDR=10.255.255.1
OMR_ADDR=10.255.255.2
VPS_IFACE=gt-tun0

Roken120, 2018-12-03:
Do you speack french ?

Ysurac, 2018-12-03:
Can be good if glorytun tcp is used.
What is the result of `ip addr show omr-6in4` on the VPS and the omrvpn ipv4 address on OpenMPTCProuter ?
Yes I speack french.

Roken120, 2018-12-03:
root@vps-40416:~# ip addr show omr-6in4
6: omr-6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1480 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.255.1 peer 10.255.255.2
    inet6 fe80::aff:ff01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/64 scope link
       valid_lft forever preferred_lft forever
the omrvpn ipv4 address on OpenMPTCProuter ?
Tu peut parler francais sa seriez plus simple car mon anglais est pas très bon

Ysurac, 2018-12-03:
Oui, j'avais remarqué ;)
L'IPv4 de l'interface omrvpn, disponible dans l'interface graphique de la partie routeur.

Roken120, 2018-12-03:
Ip local du routeur : 192.168.10.1
Ip distant du serveur : 188.213.26.3
Si tu veut on peut se parler via discord ou autre plateforme pour que je te fassent un partager d'écran :)

Ysurac, 2018-12-03:
Non j'ai pas discord ou autre ;)
Dans le menu Réseau->Interfaces tu as normalement une interface nommée OMRVPN.
Sinon par SSH sur le routeur `ip addr show 6in4-omr6in4`

Roken120, 2018-12-03:
ip addr show 6in4-omr6in4
11: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.255.28 peer 10.255.255.1

Ysurac, 2018-12-03:
ok, j'ai agrandi la plage de DHCP mais pas celle de test du service pour le 6in4...
J'ai mis à jour le script pour le VPS ce qui va résoudre ce soucis.

Ysurac, 2018-12-03:
J'ai eu des soucis avec Firstheberg dernièrement alors que j'en avais pas précédemment...
Actuellement j'utilise Scaleway.

Roken120, 2018-12-03:
C'est du 200mbps upload et download (400mbps ?)

Roken120, 2018-12-03:
Et c'est localiser en france ?

Ysurac, 2018-12-04:
C'est Online, groupe Iliad, donc Free.
Les Starts permettent une bande passante de 100 à 400Mb/s selon ce qu'on paye (dans les 2 sens donc).
Par contre tous les ports pour enovyer des mails par SMTP sont bloqués (25,465,587).

closed 2018-12-04 by Ysurac