Ysurac/openmptcprouter #238: USB WiFi Adapter randomly disassociating/disconnecting from WiFi AP

reedog117, 2018-12-05:
## Expected Behavior

WiFi adapter should stay connected to the AP.

## Actual Behavior

When having multiple of the same brand/type USB WiFi Adapter, the adapters randomly disconnect

## Steps to Reproduce the Problem

This problem occurs with an IPtime-branded USB WiFi 2.4Ghz adapter. I currently have 2 installed in a RPi3. The first one reliably stays connected to the AP. The second one connects for a few seconds and then disconnects. Seems like some sort of kernel error causing the frequent disconnects/reconnects.

```
[ 772.044915] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  773.205286] wlan0: authenticate with f2:99:b6:53:41:03
[  773.223753] wlan0: send auth to f2:99:b6:53:41:03 (try 1/3)
[  773.232726] wlan0: authenticated
[  773.244135] wlan0: associate with f2:99:b6:53:41:03 (try 1/3)
[  773.272423] wlan0: RX AssocResp from f2:99:b6:53:41:03 (capab=0x411 status=0 aid=1)
[  773.283608] usb 1-1.3: rtl8xxxu_bss_info_changed: HT supported
[  773.292056] wlan0: associated
[  773.296633] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  773.296633] 
[  773.343276] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by f2:99:b6:53:41:03
[  783.297673] wlan0: deauthenticating from f2:99:b6:53:41:03 by local choice (Reason: 3=DEAUTH_LEAVING)
[  794.204582] wlan0: authenticate with f2:99:b6:53:41:03
[  794.223111] wlan0: send auth to f2:99:b6:53:41:03 (try 1/3)
[  794.433872] wlan0: send auth to f2:99:b6:53:41:03 (try 2/3)
[  794.653929] wlan0: send auth to f2:99:b6:53:41:03 (try 3/3)
[  794.863869] wlan0: authentication with f2:99:b6:53:41:03 timed out
[  796.814470] wlan0: authenticate with f2:99:b6:53:41:03
[  796.832860] wlan0: send auth to f2:99:b6:53:41:03 (try 1/3)
[  797.043864] wlan0: send auth to f2:99:b6:53:41:03 (try 2/3)
[  797.253995] wlan0: send auth to f2:99:b6:53:41:03 (try 3/3)
[  797.463862] wlan0: authentication with f2:99:b6:53:41:03 timed out
```

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.42.7 r7405-06a20afb34 / LuCI Master (git-18.336.25424-6b2eef9)
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter platform: RPI3/x86_64

Ysurac, 2018-12-05:
Nothing I can really do about that: the auth is sent to f2:99:b6:53:41:03 but this timed out, so no answer from this device.

reedog117, 2018-12-06:
Not sure why it's timing out when it's working with a different USB Wifi adapter (mt8xxx based). I'm wondering if there are any updates to the USB WiFi module support for this type of usb adapter (rtl8xxx).

Ysurac, 2018-12-06:
This use latest driver available in current Linux 4.14 release.

reedog117, 2018-12-06:
After a closer look at the logs it looks like this is the real problem.

```
[  773.296633] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
```

closed 2019-02-20 by Ysurac