Ysurac/openmptcprouter #240: IPV6 Show

Dragonox77, 2018-12-09:
Hi,
First of all, thank you for your work, this is INSANE.

I've just updated openmptcprouter to the new version and since this patch, I don't understand why when I go to this website: http://www.mon-ip.com/ I see my ipv6 (not even my vps ipv6) and not my ipv4...
 Is that normal ?

This is a fresh install, i've just enter my 2 routers and my vps key (then ShadowSocks and Glorytun keys initialised themself)

## Specifications

  - OpenMPTCProuter version: v0.43
  - OpenMPTCProuter VPS version: v0.43
  - OpenMPTCProuter platform: RPI3/x86_64

Ysurac, 2018-12-10:
If you see IPv6 of one of your WAN, then you have to configure the modem to not send IPv6 route announce.

Dragonox77, 2018-12-10:
Thank you, i've deactivated ipv6 from my adsl router and now work perfectly fine 😄

closed 2018-12-10 by Dragonox77