Ysurac/openmptcprouter #242: OMR-Bypass

Ajowan84, 2018-12-11:
Hello,

Where can I find information about OMR-Bypass ?
I would like to know if I can get bypass OMR by going directly through a WAN ?

Thank you

## Specifications

  - OpenMPTCProuter version: 0.43
  - OpenMPTCProuter VPS version: 0.77
  - OpenMPTCProuter platform: x86_64 on VMware

Ysurac, 2018-12-12:
Source code of OMR-Bypass is available here: https://github.com/Ysurac/openmptcprouter-feeds/tree/master/luci-app-omr-bypass
Yes with OMR Bypass you can choose to output directly by a specific WAN for each domain, ip, protocol, source mac address,...

Ysurac, 2018-12-12:
Seems to be working for me. Try to update to latest OpenMPTCProuter release.

Ajowan84, 2018-12-12:
Same thing with 0.43.3
In my 4G interface, I have the following error message: "Multipath seems to be blocked on the connection"
That could be the cause?

Ysurac, 2018-12-12:
This could be a cause of not aggregation but it's all.
It's working if you use "default" interface in omr-bypass ?

Ajowan84, 2018-12-12:
"default" not working.
Maybe a problem with the configuration of my VM under VMware ...

closed 2018-12-21 by Ysurac