Ysurac/openmptcprouter #243: ERROR: block all requests from...

cron410, 2018-12-12:
## Expected Behavior
Shadowsocks accept connections if key is correct
  In English please.

## Actual Behavior

I obfuscated my home connection Public IP with [LTE Connection 1]

```
ubuntu@ubunturouter:~$ sudo systemctl status shadowsocks-libev-server@config
● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; indirect; vendor preset: enabled)
   Active: active (running) since Wed 2018-12-12 10:02:50 EST; 3h 57min ago
     Docs: man:ss-server(1)
 Main PID: 3394 (ss-server)
   CGroup: /system.slice/system-shadowsocks\x2dlibev\x2dserver.slice/shadowsocks-libev-server@config.service
           └─3394 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json --no-delay

Dec 12 13:58:11 ubunturouter ss-server[3394]:  2018-12-12 13:58:11 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:58:31 ubunturouter ss-server[3394]:  2018-12-12 13:58:31 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:58:41 ubunturouter ss-server[3394]:  2018-12-12 13:58:41 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:58:52 ubunturouter ss-server[3394]:  2018-12-12 13:58:52 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:02 ubunturouter ss-server[3394]:  2018-12-12 13:59:02 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:12 ubunturouter ss-server[3394]:  2018-12-12 13:59:12 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:32 ubunturouter ss-server[3394]:  2018-12-12 13:59:32 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:42 ubunturouter ss-server[3394]:  2018-12-12 13:59:42 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:52 ubunturouter ss-server[3394]:  2018-12-12 13:59:52 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:53 ubunturouter ss-server[3394]:  2018-12-12 13:59:53 ERROR: block all requests from [LTE Connection 1]

ubuntu@ubunturouter:~$ sudo systemctl status shadowsocks-libev-server@config2
● shadowsocks-libev-server@config2.service - Shadowsocks-Libev Custom Server Service for config2
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; indirect; vendor preset: enabled)
   Active: active (running) since Wed 2018-12-12 10:02:48 EST; 3h 57min ago
     Docs: man:ss-server(1)
 Main PID: 3390 (ss-server)
   CGroup: /system.slice/system-shadowsocks\x2dlibev\x2dserver.slice/shadowsocks-libev-server@config2.service
           └─3390 /usr/bin/ss-server -c /etc/shadowsocks-libev/config2.json --no-delay

Dec 12 13:56:29 ubunturouter ss-server[3390]:  2018-12-12 13:56:29 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:56:59 ubunturouter ss-server[3390]:  2018-12-12 13:56:59 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:57:10 ubunturouter ss-server[3390]:  2018-12-12 13:57:10 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:57:20 ubunturouter ss-server[3390]:  2018-12-12 13:57:20 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:57:30 ubunturouter ss-server[3390]:  2018-12-12 13:57:30 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:58:01 ubunturouter ss-server[3390]:  2018-12-12 13:58:01 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:58:21 ubunturouter ss-server[3390]:  2018-12-12 13:58:21 ERROR: block all requests from [LTE Connection 1]
Dec 12 13:59:22 ubunturouter ss-server[3390]:  2018-12-12 13:59:22 ERROR: block all requests from [LTE Connection 1]
Dec 12 14:00:04 ubunturouter ss-server[3390]:  2018-12-12 14:00:04 ERROR: block all requests from [LTE Connection 1]
Dec 12 14:00:14 ubunturouter ss-server[3390]:  2018-12-12 14:00:14 ERROR: block all requests from [LTE Connection 1]
```

## Steps to Reproduce the Problem

  1. Don't know what caused the problem.

## Specifications

  - OpenMPTCProuter version:  v0.42.7 r7405-06a20afb34
  - OpenMPTCProuter VPS version: 0.77
  - OpenMPTCProuter platform: x86_64

cron410, 2018-12-12:
I solved it by restarting the shadowsocks services, just wanted to document here. Should I submit an issue to Shadowsocks team?

Ysurac, 2018-12-12:
No. ss-server block connections after some handshakes failure. 
It's more a problem on OpenMPTCProuter side, I will try to reproduce the issue and fix this.

closed 2019-03-13 by Ysurac