Ysurac/openmptcprouter #244: MPTCP Change master interface...

cron410, 2018-12-12:
## Expected Behavior

Not sure what correct behavior is

## Actual Behavior

From OpenWRT SystemLog

```
Wed Dec 12 19:05:55 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan3 metric 8
Wed Dec 12 19:05:55 2018 user.notice post-tracking-post-tracking: Multipath wan3 switched to on
Wed Dec 12 19:06:39 2018 user.notice post-tracking-post-tracking: Change master interface from wan2 to wan3
Wed Dec 12 19:06:44 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan3 metric 1
Wed Dec 12 19:06:57 2018 user.notice post-tracking-post-tracking: Change master interface from wan3 to wan2
Wed Dec 12 19:07:08 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan2 metric 1
Wed Dec 12 19:12:16 2018 user.notice post-tracking-post-tracking: Change master interface from wan2 to wan3
Wed Dec 12 19:12:26 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan3 metric 1
Wed Dec 12 19:12:35 2018 user.notice post-tracking-post-tracking: Change master interface from wan3 to wan2
Wed Dec 12 19:12:41 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan2 metric 1
Wed Dec 12 19:16:38 2018 user.notice post-tracking-post-tracking: Change master interface from wan2 to wan3
Wed Dec 12 19:16:46 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan3 metric 1
Wed Dec 12 19:16:58 2018 user.notice post-tracking-post-tracking: Change master interface from wan3 to wan2
Wed Dec 12 19:17:06 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan2 metric 1
Wed Dec 12 19:23:15 2018 user.notice post-tracking-post-tracking: Change master interface from wan2 to wan3
Wed Dec 12 19:23:22 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan3 metric 1
Wed Dec 12 19:23:40 2018 user.notice post-tracking-post-tracking: Change master interface from wan3 to wan2
Wed Dec 12 19:23:52 2018 user.notice post-tracking-post-tracking: Set server **[VPS IP]** route via wan2 metric 1
```

## Steps to Reproduce the Problem

  1. Be a n00b,
  2. Fail to understand how MPTCP works :)

## Specifications

- OpenMPTCProuter version: v0.42.7 r7405-06a20afb34
- OpenMPTCProuter VPS version: 0.77
- OpenMPTCProuter platform: x86_64

Ysurac, 2018-12-12:
Update to latest version.
By default master is dynamic: connection with best RTT is used as master (if new RTT < 2*current master RTT). This can be disabled in *OpenMPTCProuter* menu, *Advanced settings* tab.

Vinc32du44, 2018-12-14:
Same for me it changes constantly from one to the other.

Ysurac, 2018-12-14:
For now you can disable dynamic master, and a new mode will be added in next release.

cron410, 2018-12-17:
I updated to latest version on both VPS and router, now the connection is very unstable. Status page bounces between the following messages for the VPS box:
```
No WAN IP Address detected in less than 1 second
No VPS IP address, No WAN IP address
Can't get public IP address from ShadowSocks
```

[image]

Ysurac, 2018-12-17:
Did you try to reboot OpenMPTCProuter ?

Ysurac, 2018-12-18:
new mode is available in latest OpenMPTCProuter, to change master only after network changes.

cron410, 2018-12-29:
You were correct, changing master interface is better Dynamic. When I had it set to a bad interface, the system incorrectly performed Direct Output and all traffic was going out the bad interface. No attempt was made to tunnel traffic to the VPS.

ByteArrayException, 2022-09-21:
@Ysurac Is it possible that when using the master dynamically, the TCP connections are interrupted briefly when switching? I have recently observed this often during downloads.