Ysurac/openmptcprouter #245: Initial configuration: How to change default address?

lars18th, 2018-12-13:
Hi,

When installing a new OpenMPTCProuter the IP address is by default `192.168.100.1` (see https://github.com/Ysurac/openmptcprouter/wiki/OpenMPTCProuter-configuration).

But when installing it in a virtual environment (or with physical devices too), in the first start we can log-in to the terminal. Why no put here some script (aka wizard) to change the default IP ?

With this, will be more easy to complete the setup, _without change anything in the current environment_.

I hope you consider it.
Regards.

Ysurac, 2018-12-13:
When you have terminal access you can edit **/etc/config/network**, I'm not sure using a script will make this easier ?

lars18th, 2018-12-13:
> When you have terminal access you can edit **/etc/config/network**, I'm not sure using a script will make this easier ?

Yes! That's I do it!
However, I suggest:

1. Document it (almost) in the Wiki.
2. One simple _shell_ script for initial configuration can be useful.

Regards.

cron410, 2018-12-13:
You can also use `uci` to set IP addresses. Afterall, this is OpenWRT at the core. https://openwrt.org/docs/guide-user/network/lede_as_routerdevice

lars18th, 2018-12-13:
> You can also use `uci` to set IP addresses. Afterall, this is OpenWRT at the core. https://openwrt.org/docs/guide-user/network/lede_as_routerdevice

Great idea! Thank you @cron410 .

Please, @Ysurac ... can you insert this info in the Wiki for people that start the setup from the console?
```
uci set network.lan.ipaddr='your-static-ip-address'
uci commit && service network restart
reboot
```
:wink:

closed 2018-12-21 by Ysurac