Ysurac/openmptcprouter #249: Help: OpenVPN setup

lars18th, 2018-12-13:
Hi,

In the OMR Luci GUI I don't see any option related to OpenVPN. However, the package is installed by default.

Any guide to setup it?

Thank you!

Ysurac, 2018-12-13:
OpenVPN TCP is used by default and it's configured using the wizard. There is a luci interface for OpenVPN available in packages.

lars18th, 2018-12-13:
Thank you!

Now I have a "similar" question: Instead of use OpenVPN, I can use GloryTUN in TCP mode, and the MPTCP will work on top of it?

Ysurac, 2018-12-13:
Yes, it's why Glorytun TCP is the default setting ;)

lars18th, 2018-12-13:
> Yes, it's why Glorytun TCP is the default setting ;)

OK. So for my environment I only need to configure two Glorytun TCP connections, or three?

- WAN1: Master, direct public IP (adsl)
- WAN2: Secondary, TCP only [HTTP proxy] (cable)
- WAN3: Secondary, TCP only [HTTP proxy] (wifi)

I know that I need to do by myself the HTTP proxy connection for Glorytun. But in this particular scenario, I need to setup WAN1+VPN2+VPN3, or I need to configure Gloryrun over _all_ three interfaces?

Please, help me to understand it.

Ysurac, 2018-12-13:
In your configuration this seems difficult... I never considered this type of configuration...
OpenMPTCProuter use one VPN to aggregate all connections, and MPTCP use the best route(s) to the server.
I have really no idea how I can manage this... I would need to make some tests with HTTP proxy and not sure this can work.

lars18th, 2018-12-13:
Hi,

Regarding the HTTP proxy, I'm searching for best solution. The goal is simple: make Glorytun TCP to run over a TCP tunnel. I feel some tool available inside OpenWRT can help.

My problem is another: When WAN2 & WAN3 are running _over_ a Glorytun VPN. In this case, it's possible to execute MPTCP on-top of WAN1, VPN2 and VPN3 and aggregate the three connections?

lars18th, 2018-12-13:
Hi @Ysurac ,

Please describe if this is true:

- OMR does aggregation _only_ with _just_ one of these:
 1. MultiPath-TCP when two ore more WAN connections support it.
 2. OR with the _internal_ Glorytun multipath support (that I suspect only works with UDP).
 3. OR with the _internal_ multipath support in MLVPN.

If that's true, then yes... my configuration is difficult!
- I test it (for a lot of time) MLVPN, and it doesn't work great.
- Glorytun I feel doesn't do multipath over TCP, so it's out-of-scope.
- Only MPTCP can work, however in this case you need to support it **over** (on-top) of VPN connections. With the limitation of the TCP-over-TCP poor performance when the VPN runs over TCP.

What you think?

Ysurac, 2018-12-13:
OMR does agregation with MPTCP supported on one ore more connections AND with Glorytun TCP VPN (for UDP traffic only) AND with Shadowsocks (for TCP traffic only).
MPTCP is Multipath TCP, it's a protocol compatible with TCP and any TCP application can work over MPTCP. Glorytun TCP VPN is only here to make a VPN between OMR and VPS so flow can be aggregated on VPS end.
So this seems difficult to use with HTTP proxy.

lars18th, 2018-12-13:
Hi @Ysurac ,

I know a lot about MPTCP. So, this isn't the question.
And I know too about MLVPN (that's not very efficient).

I don't know _how_ with ShadowSocks you're doing multipath. I feel you only rely in MPTCP for _any_ upper protocol that uses TCP packets. That's right?

closed 2018-12-26 by Ysurac