Ysurac/openmptcprouter #251: Trouble connection YouTube App on the phone to Roku

dnwk, 2018-12-13:
## Expected Behavior

You could project a youtube video on your phone to Roku or Chromecast.

## Actual Behavior
Since start using OpenMPTC, youtube app would recognize there is a Roku on the network but could never connect to it. I don't know what kind of protocol involved. Is the router blocking some kind of package? Or because the router is running as a VM may cause some problem? 

## Steps to Reproduce the Problem

  1. In youtube app on your phone, use project bottom and choose roku player
  2. It will listed the name of roku player. Click on it.
  3. App will trying to connect to Roku player but ultimately timed out. 

## Specifications

  - OpenMPTCProuter version: (Last is not a version)  2018/12/11 
  - OpenMPTCProuter VPS version: (Last is not a version)  0.77
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) X86_64

Ysurac, 2018-12-13:
This may be the same problem as https://github.com/Ysurac/openmptcprouter/issues/160#issuecomment-420547573 with hard coded Google DNS. Even if I don't understand why this would change anything ;)
I don't have a Roku or Chromecast to test...

Vinc32du44, 2018-12-14:
For my case on youtube PC the picture is jerky since the last update.

dnwk, 2018-12-14:
> This may be the same problem as [#160 (comment)](https://github.com/Ysurac/openmptcprouter/issues/160#issuecomment-420547573) with hard coded Google DNS. Even if I don't understand why this would change anything ;)
> I don't have a Roku or Chromecast to test...

Not exactly sure what's the solution in #160

dnwk, 2019-02-07:
Problem seem to solve it self. It seems related to my previous virtual machine setup. Could be related to how virtual machine handles network bridge.

closed 2019-02-07 by Ysurac