Ysurac/openmptcprouter #252: Aggregate Satellite dishes

MrTheSkream, 2018-12-14:
Hello,
I currently have a full installation of openmptcprouter, server + client, and i would like to help with that aggregate two satellite dishes that have a latency of 700ms.
However there is a warning.

Has anyone ever managed to do it?

## Steps to Reproduce the Problem

  1. Two WAN satellite dishes
  2. One Raspberry pi client
  3. One Dedicated Server

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.43.3 r7407-1cd945ea22 / LuCI Master (git-18.345.50473-b920e86)
  - OpenMPTCProuter VPS version: debian9-x86_64
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

Ysurac, 2018-12-14:
This should work without real problems. You can have a warning for each wan that server take more than 1 second to answer, but it's only a warning about slow latency.

MrTheSkream, 2018-12-14:
Aggregation is not done.
It seems to me that TCP is sensitive to latency so I wonder if it does not come from there.

Ysurac, 2018-12-14:
Can you make a capture of your status page ?

Ysurac, 2018-12-14:
Configuration seems to be ok. You can try latest OpenMPTCProuter v0.44 (release a few seconds ago).
I will try to simulate slow latency connection and see if I can solve issues.

Ysurac, 2018-12-14:
I'm able to reproduce the issue. I will try to fix this ASAP.

MrTheSkream, 2018-12-14:
Okey thanx

MrTheSkream, 2018-12-17:
@Ysurac J'ai vu que tu étais français, donc si cela te dérange pas on peut parler dans cette langue pour une meilleur compréhension.

J'ai testé OpenMPTCProuter v0.44 et cela ne corrige pas le problème.

MrTheSkream, 2019-01-07:
Hey, some news ?

max4186, 2020-05-31:
@Ysurac I wondered if you ever got anywhere with this, I plan on bonding my fixed wireless internet (50/20) with a 4G signal which is provided by Satellite (600ms but it is a steady 10/4 connection). Do you think this will work? Any tips?

Ysurac, 2020-06-01:
I can make some simulations, but for that I would need the result of a ping via each connection (I need latency, jitter and packet lost).
From last test I've done, aggregation was really bad when ping is over 500ms. Maybe I can try to improve that...

kevin39, 2020-06-01:
I think it's a lot too high to get a working aggregation... 
I think, if all the connections are slow this may work but mixing fast and (very) connection will decrease performance... No ?

I was having sat connection 1 year ago and never found a good way to have aggregation.

Ysurac, 2020-06-01:
If all connection have high latency this should be possible to make it work.

max4186, 2020-06-01:
The main connection has around 40ms and 2-5 jitter. The satellite connection is about 670ms and a jitter of 40-80. Would that work?

Adorfer, 2020-09-30:
"high latency" is not an problem if all(!) connections experience the same amount of latency (within a close range of a e.g. 50-100ms: 700ms/720ms/770ms). But if the latency span is big (40ms/600ms) the MPTCP will probably not aggregate without tweaking. 
(and if you do, everything(!) going over the tunnel will have 600ms LAG, so perhaps you have to OMRbypass a lot of stuff)

closed 2021-01-04 by github-actions[bot]