Ysurac/openmptcprouter #257: VPS pool

mycanaletto, 2018-12-16:
Hello,
Do you think it will be possible to "pool" the VPS side to allow several MPTCP routers to use it for outgoing frafic? What happens if you use keys on multiple routers?

Ysurac, 2018-12-16:
no. It will not work. Multiple client on VPS will be supported, but not for now.

closed 2018-12-16 by Ysurac

mycanaletto, 2018-12-16:
Merci pour l’info et bravo pour le projet 😊

https://www.canaletto.fr/post/mptcp-vs-dual-wan

mycanaletto, 2018-12-26:
Bonjour Yannick,

Avec deux VDSL, j’ai toujours le up qui s’effondre rapidement (le VPS est dans une VM dans un datacenter avec 1 Gbit/s)

Une idée ?

Merci, lionel

[cid:image001.png@01D49D68.80738140]

mycanaletto, 2019-06-19:
Bonjour,
Comment générer de nouvelles clés sur le VPS ? (mon idée étant de dupliquer la VM du VPS en changeant son IP publique plutôt que de la recréer...).