Ysurac/openmptcprouter #258: Serveur Ovh Acces exterieur

soken35, 2018-12-16:
Bonjour.

J'ai installé openmptcp sur mon serveur ovh sans souci et le routeur sous esxi.
Tout fonctionne au parfaitement, l’agrégation fonctionne etc.

Cependant, je n'ai plus accès a mon serveur OVH depuis internet sans être connecté a mon réseau local.

Savez vous comment remédier à ce problème ? 

Merci beaucoup

  - OpenMPTCProuter version: 4.14.88
  - OpenMPTCProuter VPS version: 4.14.79
  - OpenMPTCProuter platform: ESXI x86_64)

kamcio2603, 2018-12-16:
*English please*. What is problem with that sentence?!

soken35, 2018-12-16:
Hello.

I installed openmptcp on my ovh server without worry and the router under esxi.
Everything works perfectly, aggregation works etc.

However, I no longer have access to my OVH server from the internet without being connected to my local network.

Do you know how to fix this problem?

Sorry for the google translation

kamcio2603, 2018-12-16:
"no longer have access to my OVH server from the internet" - what do you mean? Cant connect to SSH ? If yes, try port 65222

ssh IP.IP.IP -p 65222 -l LOGIN

soken35, 2018-12-16:
I do not have ssh access. I can not open a site (port80) connection refused, more access for example to the plex server (port 32400). If I connect from home, no problem it works.

kamcio2603, 2018-12-16:
Yep, its because mptcprouter new firewall rules.Defaults to block ALL not defined. You need to edit rules from /etc/shorewall/rules with file editor you like (vim,nano mcedit etc..)

soken35, 2018-12-16:
That is what I thought. I do not master shorewall. I do not see what is the rule to edit. can you guide me? thank you so much for your responsiveness

Ysurac, 2018-12-17:
It's not really difficult, all rules are in **/etc/shorewall/rules** for IPv4 and **/etc/shorewall6/rules** for IPv6.
Then there is already the SSH example.

soken35, 2018-12-17:
Here's what I added to the rules file:

 ACCEPT        net     $FW       tcp        80

But I do not have access.
I have good access when I cut the firewall

soken35, 2018-12-17:
In local I have access to the site with the address 10.255.255.1 but not with the ip of the server

kamcio2603, 2018-12-17:
After editing shorewall rules did you do "systemctl restart shorewall"?

soken35, 2018-12-17:
Yes.

soken35, 2018-12-17:
So it's very strange, I did not do anything more but it works. I did not add any rules. If it changes, I'll let you know.

thanks again

soken35, 2018-12-17:
Well, finally it does not work anymore!

kamcio2603, 2018-12-17:
Tested on my VPS. Works great (80 port open) - website is reachable. Maybe syntax error in shorewall?

soken35, 2018-12-17:
``#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
##################################################################################################################################################$
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TI$
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

ACCEPT          net             $FW             tcp     80
ACCEPT          net             $FW             tcp     21

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999

Ysurac, 2018-12-17:
You should open ports after ```Invalid(DROP)   net             all             tcp```

soken35, 2018-12-17:
Like this ?
It still does not work

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
ACCEPT          net             $FW             tcp     80
#

kamcio2603, 2018-12-17:
Hey, whats your name of interface? eth0 or something else? (enp...)

kamcio2603, 2018-12-17:
...and check your params.net : 

root@debian:/etc/shorewall# cat params.net
NET_IFACE=eth0

my vps interface is eth0, so all works great.

Ysurac, 2018-12-17:
Should work.
Make sure your server really listen at port 80 and doesn't redirect to 443.

soken35, 2018-12-17:
> ...and check your params.net :
> 
> root@debian:/etc/shorewall# cat params.net
> NET_IFACE=eth0
> 
> my vps interface is eth0, so all works great.

```
root@PrOxyGeN:/etc/shorewall# cat params.net
NET_IFACE=eno1root@PrOxyGeN:/etc/shorewall#
```

soken35, 2018-12-17:
> Should work.
> Make sure your server really listen at port 80 and doesn't redirect to 443.

My file /etc/apache2/ports.conf

```
Listen 80
<IfModule ssl_module>
        Listen 443

```

kamcio2603, 2018-12-17:
netstat -plnt | grep ':80'

and check if it "LISTEN". Your problem seems weird.

soken35, 2018-12-17:
indeed, the installation of OpenMPTCP has changed the configuration of the server.

`netstat -plnt | grep ':80'
tcp        0      0 0.0.0.0:8069            0.0.0.0:*               LISTEN      571/python3
tcp6       0      0 :::8080                 :::*                    LISTEN      992/java
tcp6       0      0 :::80                   :::*                    LISTEN      1035/apache2
tcp6       0      0 :::8096                 :::*                    LISTEN      939/mono-sgen
tcp6       0      0 :::8065                 :::*                    LISTEN      574/mattermost
tcp6       0      0 127.0.0.1:8005          :::*                    LISTEN      992/java
`

soken35, 2018-12-17:
Why since the installation of openmptcp I can not find the files anymore /etc/network/interfaces ?

Ysurac, 2018-12-17:
The script doesn't do anything to this file.

soken35, 2018-12-17:
ok it has passed something else, but I have no more file interfaces. But all interfaces are still mounted.

`eno1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet **ip-serv**  netmask 255.255.255.0  broadcast **ip-serv**
        inet6 2001:41d0:303:5a4b::  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::a6bf:1ff:fe2b:862c  prefixlen 64  scopeid 0x20<link>
        ether a4:bf:01:2b:86:2c  txqueuelen 1000  (Ethernet)
        RX packets 4558470  bytes 3905899037 (3.6 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 5576695  bytes 6110002684 (5.6 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0xa2b00000-a2bfffff

gt-tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.255.255.1  netmask 255.255.255.0  destination 10.255.255.1
        inet6 fe80::c327:af94:8688:ff00  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 87198  bytes 7489352 (7.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 139436  bytes 152646425 (145.5 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

gt-udp-tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet 10.255.254.1  netmask 255.255.255.0  destination 10.255.254.1
        inet6 fe80::46f:c2b8:3a85:6c68  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 1972 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Boucle locale)
        RX packets 41409  bytes 8288250 (7.9 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 41409  bytes 8288250 (7.9 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

omr-6in4: flags=209<UP,POINTOPOINT,RUNNING,NOARP>  mtu 1480
        inet6 fe80::aff:ff01  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::a00:1  prefixlen 64  scopeid 0x20<link>
        sit  txqueuelen 1000  (IPv6-dans-IPv4)
        RX packets 405  bytes 37383 (36.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1463  bytes 264154 (257.9 KiB)
`

soken35, 2018-12-17:
Here is my file /etc/shorewall/interfaces

# For information about entries in this file, type "man shorewall-interfaces"
###############################################################################
?FORMAT 2
###############################################################################
#ZONE   INTERFACE       OPTIONS
net     $NET_IFACE      dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0
vpn     gt-tun0         nosmurfs,routefilter,logmartians,tcpflags
vpn     gt-udp-tun0     nosmurfs,routefilter,logmartians,tcpflags
vpn     mlvpn0          nosmurfs,routefilter,logmartians,tcpflags
vpn     tun0            nosmurfs,routefilter,logmartians,tcpflags

```

Ysurac, 2018-12-17:
Your webserver only listen port 80 on IPv6, so it's ```/etc/shorewall6/rules```

soken35, 2018-12-17:
Yes, same as showrewall / rules

```
#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
ACCEPT          net             $FW             tcp     80
#
```

Ysurac, 2018-12-17:
And you also restarted shorewall6 ?

soken35, 2018-12-17:
Yes.

soken35, 2018-12-17:
Here is my files /etc/shorewall/interfaces 

```
#ZONE   INTERFACE       OPTIONS
net     $NET_IFACE      dhcp,tcpflags,routefilter,nosmurfs,logmartians,sourceroute=0
vpn     gt-tun0         nosmurfs,routefilter,logmartians,tcpflags
vpn     gt-udp-tun0     nosmurfs,routefilter,logmartians,tcpflags
vpn     mlvpn0          nosmurfs,routefilter,logmartians,tcpflags
vpn     tun0            nosmurfs,routefilter,logmartians,tcpflags

and my file /etc/showrewall6/interfaces/

```
#ZONE   INTERFACE       OPTIONS
net     eno1            dhcp,tcpflags,rpfilter,forward=1
vpn     omr-6in4        tcpflags,forward=1

```
```

soken35, 2018-12-17:
> ...and check your params.net :
> 
> root@debian:/etc/shorewall# cat params.net
> NET_IFACE=eth0
> 
> my vps interface is eth0, so all works great.

Do you have the same content daqns your files?

soken35, 2018-12-17:
I finally found the problem.
In OpenMtcprouter you have to uncheck this.
[image]
All the rules create on the server are not taken into account since all the ports are redirected towards the router.
Thanks again to all for your help and your responsiveness.

Ysurac, 2018-12-18:
It's disabled by default, but yes this make all ports redirected to OpenMPTCProuter.

closed 2018-12-18 by Ysurac