Ysurac/openmptcprouter #259: Gateway Down

odcw, 2018-12-17:
## Expected Behavior

## Actual Behavior

Gateway Down

## Steps to Reproduce the Problem

  1)  wan1  (master) connected directly to the Box Ip static => it's works
  2) eth1 (backup) 4G with usb huawei E3372 ip dhcp  I get ip but I've a message Gateway down => doesn't work

## Specifications

  - OpenMPTCProuter version: (0.44.2)
  - OpenMPTCProuter VPS version: (0.77)
  - OpenMPTCProuter platform: (RPI3/x86_64)

Ysurac, 2018-12-17:
I need the result of ```uci show network```, ```ubus call network.interface.wan2``` and  ```ubus call network.interface.wan2_4``` (replace wan2 by the name you give to the interface).

odcw, 2018-12-17:
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.globals=globals
network.globals.ula_prefix='fdc7:3110:804c::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_syn_retries='5'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.metric='2048'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='b8:27:eb:78:89:56'
network.lan.ipaddr='192.168.1.1'
network.lan.gateway='192.168.1.254'
network.lan.dns='192.168.1.254'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.macaddr='da:a2:17:57:cb:fe'
network.wan1.ipaddr='192.168.1.253'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.1.254'
network.wan1.dns='192.168.1.254'
network.wan1.multipath='master'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.type='macvlan'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth0'
network.wan2.macaddr='6e:92:d1:42:f5:fe'
network.wan2.ipaddr='192.168.0.2'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.8.1'
network.wan2.dns='192.168.8.1'
network.wan2.multipath='off'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='da:a2:17:57:cb:fe'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth0'
network.wan2_dev.macaddr='6e:92:d1:42:f5:fe'
network.@route6[0]=route6
network.@route6[0].interface='lan'
network.@route6[0].target='::/0'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
network.omr6in4.ipaddr='10.255.255.28'
network.4G=interface
network.4G.ifname='eth1'
network.4G.proto='static'
network.4G.ipaddr='192.168.8.2'
network.4G.netmask='255.255.255.0'
network.4G.gateway='192.168.8.1'
network.4G.dns='192.168.8.1'
network.4G.macaddr='0c:5b:8f:27:9a:64'
network.4G.multipath='on'
network.wwan=interface
network.wwan.proto='static'
network.wwan.multipath='off'
network.wwan.ifname='eth0'
network.wwan.macaddr='b8:27:eb:78:89:56'
network.tplink=interface
network.tplink.proto='dhcp'
network.tplink.multipath='on'
root@OpenMPTCProuter:~#

odcw, 2018-12-17:
root@OpenMPTCProuter:~# ifconfig
6in4-omr6in4 Link encap:IPv6-in-IPv4
          UP POINTOPOINT RUNNING NOARP  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr B8:27:EB:78:89:56
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
          RX packets:72033 errors:0 dropped:1 overruns:0 frame:0
          TX packets:101141 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:90
          RX bytes:13837070 (13.1 MiB)  TX bytes:11725452 (11.1 MiB)

eth1      Link encap:Ethernet  HWaddr 0C:5B:8F:27:9A:64
          inet addr:192.168.8.2  Bcast:192.168.8.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:57 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10000
          RX bytes:24604 (24.0 KiB)  TX bytes:756 (756.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:79498 errors:0 dropped:0 overruns:0 frame:0
          TX packets:79498 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:11916055 (11.3 MiB)  TX bytes:11916055 (11.3 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.255.255.28  P-t-P:10.255.255.28  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:3077 errors:0 dropped:0 overruns:0 frame:0
          TX packets:692 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:90
          RX bytes:255121 (249.1 KiB)  TX bytes:58618 (57.2 KiB)

wan1      Link encap:Ethernet  HWaddr DA:A2:17:57:CB:FE
          inet addr:192.168.1.253  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4533 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1528 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10000
          RX bytes:395170 (385.9 KiB)  TX bytes:108524 (105.9 KiB)

wan2      Link encap:Ethernet  HWaddr 6E:92:D1:42:F5:FE
          inet addr:192.168.0.2  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1872 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:90
          RX bytes:156372 (152.7 KiB)  TX bytes:1188 (1.1 KiB)

wlan0     Link encap:Ethernet  HWaddr B8:27:EB:2D:DC:03
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2687 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2301 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10000
          RX bytes:224515 (219.2 KiB)  TX bytes:188746 (184.3 KiB)

Ysurac, 2018-12-17:
It's working as static and not as DHCP ?

odcw, 2018-12-17:
wan1 to box working as static and it works   eth1 as static or dhcp doesn't work

closed 2018-12-18 by odcw