Ysurac/openmptcprouter #260: SS invalid key format (bad luck ?)

PxxBxx, 2018-12-17:
## Expected Behavior

clean VPS install working

## Actual Behavior

Shadowsocks doesn't launch because of invalid key format

## Steps to Reproduce the Problem

  1. clean install
  2. invoke bad luck charm
  3.

## Details

My original key was 08iFt1latmaWg+TSfsZABodIuVypc2kOwYyun8pGtJM=
But SS was not running, and when `systemctl status` it, here was the output : 
`Dec 17 20:27:55 my-vps ss-server[26741]:  2018-12-17 20:27:55 INFO: initializing ciphers... chacha20`
`Dec 17 20:27:55 my-vps ss-server[26741]:  2018-12-17 20:27:55 ERROR: Invalid key for your chosen cipher!`
`Dec 17 20:27:55 my-vps ss-server[26741]:  2018-12-17 20:27:55 ERROR: It requires a 32-byte key encoded with URL-safe Base64`

So I entered a new base64 key, without any pluses (+) sign, and it worked.
Example : pLN7VWSPnLwlZaWIGb0HDUlJNjEl9WPmOrn_Y5k4pLo=

**=> maybe add a security in the VPS install script to avoid pluses (+) signs in the generated SS key ?**
Or the problem is elsewhere, but all 10 keys randomly generated at launch by the shadowsock binary had no pluses (+) in them.

## Specifications

  - OpenMPTCProuter version: 0.44.2
  - OpenMPTCProuter VPS version: 0.79
  - OpenMPTCProuter platform: x64

PxxBxx, 2018-12-17:
Note (important....) : I was using the same key since my first (<0.44) VPS  install.
Maybe you already have the correct security installed, and I can only blame myself.....

Ysurac, 2018-12-18:
The key used in the json config of shadowsocks doesn't have the "+" sign, it's removed by VPS script before adding in the config. Same on OpenMPTCProuter part.
Where did you put this key ?

PxxBxx, 2018-12-18:
I've had my previous key from the txt file, and I've set it (manually...) in ss/config.json **and** in openmptcprouter_config.txt
I understand my error : 
* there was a plus (+) in the _config.txt file
* the plus (+) was replaced by minus (-) in the config.json file

You have the `sed` since 7 monthes ago in the VPS install script, so It's really **my** mistake, I shouldn't have replace the json key with the txt key.

Sorry, my bad, there is no issue.

closed 2018-12-18 by PxxBxx