Ysurac/openmptcprouter #262: Version 0.44.* and 0.43 don't finish to boot on RPI3B+

Gheop, 2018-12-18:
I try 3 version of 0.44 and v0.43 on new install and no one finish to boot on my RPI3 and RPI3B+. Don't have screen to see what it do but the green led stay on.
Work great with my old v0.42.7.
Upgrade since v0.42.7 to v0.44.* don't work too with similar problem after reboot.

Gheop, 2018-12-18:
Humm, I have this problem on 2 microSD, an old Verbatim 2GB and old SanDisk Ultra 8GB class 10 HC I.

I just try v0.44.5 on new SanDisk 16GB A1 and it work like a charm on the same RPI3+ with the same peripheral devices! 

v0.42.7 work on all this SD cards.

Ysurac, 2018-12-18:
You use squashfs or ext4 image ?

Gheop, 2018-12-18:
always use ext4, never try squashfs, maybe I must ?
Don't really know the diff, I just read on wikipedia that squashfs is read-only filesystem... Can't I save my config with it ?

Ysurac, 2018-12-18:
Both should work. Squashfs sometimes give me strange result, but not the case here.
You can save the conf also with squashfs (this use also overlayfs to make it writable).

closed 2018-12-21 by Ysurac