Ysurac/openmptcprouter #263: Problem with M6replay

renaudadu, 2018-12-18:
hello,
i have some trouble with M6replay. they detect the VPN and it's not working.
it's possible to bridge the all the architecture like a modem/router with  a bridge mode ( FREEBOX in france )

## Specifications

  - OpenMPTCProuter version: (0.44.5)
  - OpenMPTCProuter VPS version: (0.79)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2018-12-18:
Bridge will not change anything.
Many VPS IPs are banned by this type of sites. You have to use omr-bypass and choose the WAN you want to use as output, but no more aggregation for this site.

PxxBxx, 2018-12-19:
I wanted to help, but I'm not at home and can't test...
If a domain (ex: 6play.fr, 6cloud.fr) is set in omr-bypass, are all subdomains (geo.6play.fr, ...) bypassed, or do we have to list all subdomains ?
For information, 6play does geoblocking on frontend and CDN side, so at least all subdomains from these 2 domains (6play.fr, 6cloud.fr) are to be bypassed.

Ysurac, 2018-12-19:
Yes all subdomains are also bypassed.

renaudadu, 2018-12-19:
hello, 
i just test and only 6play.fr et 6could.fr -> 6play not working
with geo.6play.fr -> it's working.
i think subdomains not bypassed or it's a bug

renaudadu, 2018-12-19:
no. i use my own DNS

Ysurac, 2018-12-19:
It's why subdomains are not working. They are bypassed only when a request is made on OMR DNS.

closed 2018-12-26 by Ysurac