Ysurac/openmptcprouter #264: No access to my Internat

sebdew, 2018-12-18:
## Expected Behavior

  In English please.

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

max4186, 2018-12-19:
Have you disabled DHCP on your ADSL Interface modem? Also, what is this being run on? Raspberry Pi or something else?

sebdew, 2018-12-19:
DHCP is disabled on the ADSL modems and I have manually fixed the IP of LAN and WAN interfaces on the MPTCP router. I will double check this evening that their is no DHCP enabled on my pfSense WAN interface. But I'm pretty sure it is not the case. The router is running on a Raspberry PI 2 Model B.

Ysurac, 2018-12-19:
WAN interface should not be on the same network subnet.
You used the wizard to configure them or manually ?

sebdew, 2018-12-19:
Yes, it is the case. I have a /27 for each network.

I have use the wizard to configure.
The only manual change I have done is the mask of the lan interface (move from /24 to /27).

What strange is that I'm able to ping the routers (MPTCP and ADSL) from my computer, but, I'm unable to ping something from the MPTCP router. It seems to be an iptable issue. :(

sebdew, 2018-12-19:
`root@OpenMPTCProuter:~# ip route
default via 192.168.10.1 dev wan2
1.0.0.1 via 192.168.10.1 dev wan2 src 192.168.10.3
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
192.168.10.0/27 dev wan2 proto kernel scope link src 192.168.10.3
192.168.10.32/27 dev wan1 proto kernel scope link src 192.168.10.35
`
`root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.globals=globals
network.globals.ula_prefix='fd36:e9fc:b172::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.mptcp_checksum='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_syn_retries='20'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.ip6assign='60'
network.lan.metric='2048'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='b8:27:eb:87:9b:b3'
network.lan.netmask='255.255.255.224'
network.lan.dns='192.168.100.2'
network.lan.gateway='192.168.100.2'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.macaddr='f6:ef:b2:83:8a:ff'
network.wan1.ipaddr='192.168.10.35'
network.wan1.netmask='255.255.255.224'
network.wan1.gateway='192.168.10.33'
network.wan1.label='ADSL Interface'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.type='macvlan'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth0'
network.wan2.macaddr='3e:52:c5:e7:03:7d'
network.wan2.ipaddr='192.168.10.3'
network.wan2.netmask='255.255.255.224'
network.wan2.gateway='192.168.10.1'
network.wan2.label='4G Interface'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='f6:ef:b2:83:8a:ff'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth0'
network.wan2_dev.macaddr='3e:52:c5:e7:03:7d'
network.@route6[0]=route6
network.@route6[0].interface='lan'
network.@route6[0].target='::/0'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.proto='dhcp'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2'
`

Ysurac, 2018-12-19:
All seems good. You should be able to ping gateway at least...
Can you give me ```ip addr``` ?

sebdew, 2018-12-19:
No, I'm unable to ping any gateway. :(

> root@OpenMPTCProuter:~# ip addr
> 1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope host lo
>        valid_lft forever preferred_lft forever
> 2: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UP group default qlen 90
>     link/ether b8:27:eb:87:9b:b3 brd ff:ff:ff:ff:ff:ff
>     inet 192.168.100.1/27 brd 192.168.100.31 scope global eth0
>        valid_lft forever preferred_lft forever
> 3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
>     link/sit 0.0.0.0 brd 0.0.0.0
> 4: ifb0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
>     link/ether fa:8d:45:3b:29:1b brd ff:ff:ff:ff:ff:ff
> 5: ifb1: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
>     link/ether 42:88:ce:8d:25:20 brd ff:ff:ff:ff:ff:ff
> 12: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/sit 10.255.255.2 peer 10.255.255.1
> 16: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc noqueue state UP group default qlen 10000
>     link/ether f6:ef:b2:83:8a:ff brd ff:ff:ff:ff:ff:ff
>     inet 192.168.10.35/27 brd 192.168.10.63 scope global wan1
>        valid_lft forever preferred_lft forever
> 17: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc noqueue state UP group default qlen 10000
>     link/ether 3e:52:c5:e7:03:7d brd ff:ff:ff:ff:ff:ff
>     inet 192.168.10.3/27 brd 192.168.10.31 scope global wan2
>        valid_lft forever preferred_lft forever
> 19: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UNKNOWN group default qlen 90
>     link/none
>     inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
>        valid_lft forever preferred_lft forever
>

sebdew, 2018-12-19:
But the gateway are pingable from my computer.

Ysurac, 2018-12-19:
You don't have a mac filter on the switch ? (with macvlan, we announce several mac address on one interface)
In the wizard, set "Download speed" and "Upload speed" to 0 if it's not already the case.
Else I have no more idea for now...

sebdew, 2018-12-19:
I found what was the issue.... I have remove the LAN Gateway to have the system working...

I don't know if it is a normal situation or not.

If it is normal, maybe the Gateway field should be removed from the form.

sebdew, 2018-12-19:
Anyway, it's working now. Thanks for your time and help.

closed 2018-12-19 by sebdew