Ysurac/openmptcprouter #265: OMR Bypass via ASN

max4186, 2018-12-19:
Suggestion:

Currently OMR Bypass supports Domains, IP's, MAC, Protocols and Source IP addresses - it'd be nice if you could create a bypass for all IP addresses under a specific network ASN so you wouldn't need to go through adding specific ranges for a company for example.

Ysurac, 2019-01-06:
Done in latest version, for now this is only updated when omr-bypass is restarted. I will change that in a next version.

ThibautBascoulergue, 2019-01-06:
What should we put in the field of ASN, Netflix or the corresponding ASN number?

max4186, 2019-01-06:
Works fantastically, good work Ysurac!

For the more stupid of us, you might want to add a note to not add AS infront of the number as (unless it supports having AS, I don't know).

Ysurac, 2019-11-01:
Update via cron will be added in next release. Added in develop branch.

closed 2019-11-01 by Ysurac