Ysurac/openmptcprouter #268: 30 sec lantency online multiplayer game !

Vinc32du44, 2018-12-19:
Since last update i have a lot of latency in my game ?! About 20 - 30 sec... 

All work fine but just in game...

Ysurac, 2018-12-19:
What is the VPN used ? Glorytun TCP, Glorytun UDP, OpenVPN or MLVPN ?
You have an example of game ?
Did you check the latency of the VPS ?

Vinc32du44, 2018-12-19:
OpenMPTCPRouter is on default with default VPN.
The latency of VPS is good => 22 ms
Game Destiny 2
This morning i try add server VPS but remove it right now.
When i look in my route it appear WAN1 & WAN2 on both VPS server ? (main...)
Can you explain ?

Ysurac, 2018-12-19:
I don't have this game to test... I'm testing on GTAV and Ballistic Overkill.
Not really a big problem but old route are not removed for now, should be better in next version. You can reboot router to reset route.

Vinc32du44, 2018-12-19:
Why congestion control is "Olia" since your last update ?

Ysurac, 2018-12-19:
It's still BBR by default.

Vinc32du44, 2018-12-19:
Fresh install ... all right !
I think that's other vps server stay in openmptcprouter config and reboot don't fix it !
Thanks ycarus

ThibautBascoulergue, 2018-12-22:
For me with a fresh install the congestion control is set to Olia as Vinc32du44.
I was forced to bbr on the router MPTCP and change also in the conf of the kernel on the part VPS because there too it is in default OLIA.
With a OLIA congestion i have a ping to google dns at more than 1000 ms. With BBR its normal 40~90ms. Maybe the OLIA is not compatible with full 4G connexion :)

Ysurac, 2018-12-22:
The only change is that it use congestion control on both side (if VPS was olia, then OpenMPTCProuter is set as olia). But it's BBR by default since a long time now ;)

Vinc32du44, 2018-12-22:
How can i know the congestion control use by the VPS  ?

Ysurac, 2018-12-22:
```sysctl net.ipv4.tcp_congestion_control``` on the VPS

closed 2019-02-20 by Ysurac