Ysurac/openmptcprouter #270: Router reboot itself

dnwk, 2018-12-21:
## Actual Behavior
My router will reboot itself at some random point. When I find I lose internet connection, I go to system log and I could only find system log related to starting up the router. I can't find out what trigger the reboot.

## Specifications

  - OpenMPTCProuter version: (Last is not a version)  v0.44.5 r7416-50789acf4e
  - OpenMPTCProuter VPS version: (Last is not a version)  0.79
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) x86_64

Ysurac, 2018-12-21:
Without backtrack this will be difficult to say what trigger the reboot.
But if you use x86_64 platform, that is a fully tested platform, I think it's an hardware problem.
You can try to write system log to log file here http://192.168.100.1/cgi-bin/luci/admin/system/system

dnwk, 2018-12-21:
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.050490] ACPI: (supports S0 S5)
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.050915] ACPI: Using IOAPIC for interrupt routing
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.051427] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.053061] ACPI: Enabled 16 GPEs in block 00 to 0F
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.060449] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.061014] acpi PNP0A03:00: _OSC: OS supports [Segments MSI]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.061524] acpi PNP0A03:00: _OSC failed (AE_NOT_FOUND); disabling ASPM
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.062322] acpiphp: Slot [3] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.062851] acpiphp: Slot [4] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.063370] acpiphp: Slot [5] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.063918] acpiphp: Slot [7] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.064432] acpiphp: Slot [8] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.064956] acpiphp: Slot [9] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.065469] acpiphp: Slot [10] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.065993] acpiphp: Slot [11] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.066510] acpiphp: Slot [12] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.067035] acpiphp: Slot [13] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.067552] acpiphp: Slot [14] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.068078] acpiphp: Slot [15] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.068595] acpiphp: Slot [16] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.069125] acpiphp: Slot [17] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.069641] acpiphp: Slot [18] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.070089] acpiphp: Slot [19] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.070630] acpiphp: Slot [20] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.071157] acpiphp: Slot [21] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.071682] acpiphp: Slot [22] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.072204] acpiphp: Slot [23] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.072735] acpiphp: Slot [24] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.073298] acpiphp: Slot [25] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.073853] acpiphp: Slot [26] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.074399] acpiphp: Slot [27] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.074949] acpiphp: Slot [28] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.075472] acpiphp: Slot [29] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.076011] acpiphp: Slot [30] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.076527] acpiphp: Slot [31] registered
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.076999] PCI host bridge to bus 0000:00
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.077442] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.077997] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.078536] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.079287] pci_bus 0000:00: root bus resource [mem 0x80000000-0xfebfffff window]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.080008] pci_bus 0000:00: root bus resource [bus 00-ff]
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.080570] pci 0000:00:00.0: [8086:1237] type 00 class 0x060000
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.081340] pci 0000:00:01.0: [8086:7000] type 00 class 0x060100
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.082297] pci 0000:00:01.1: [8086:7010] type 00 class 0x010180
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.092010] pci 0000:00:01.1: reg 0x20: [io  0xc0e0-0xc0ef]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.100016] pci 0000:00:01.1: legacy IDE quirk: reg 0x10: [io  0x01f0-0x01f7]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.100588] pci 0000:00:01.1: legacy IDE quirk: reg 0x14: [io  0x03f6]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.101141] pci 0000:00:01.1: legacy IDE quirk: reg 0x18: [io  0x0170-0x0177]
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.101709] pci 0000:00:01.1: legacy IDE quirk: reg 0x1c: [io  0x0376]
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.102620] pci 0000:00:01.3: [8086:7113] type 00 class 0x068000
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.103303] pci 0000:00:01.3: quirk: [io  0x0600-0x063f] claimed by PIIX4 ACPI
Thu Dec 20 18:13:00 2018 kern.info kernel: [    0.107077] pci 0000:00:01.3: quirk: [io  0x0700-0x070f] claimed by PIIX4 SMB
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.108090] pci 0000:00:02.0: [1013:00b8] type 00 class 0x030000
Thu Dec 20 18:13:00 2018 kern.debug kernel: [    0.112565] pci 0000:00:02.0: reg 0x10: [mem 0xfc000000-0xfdffffff pref]
[... cut ...]
Thu Dec 20 18:14:18 2018 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Thu Dec 20 18:14:18 2018 daemon.notice miniupnpd[4586]: shutting down MiniUPnPd
Thu Dec 20 18:14:19 2018 daemon.notice miniupnpd[10726]: HTTP listening on port 5000
Thu Dec 20 18:14:19 2018 daemon.warn miniupnpd[10726]: no HTTP IPv6 address, disabling IPv6
Thu Dec 20 18:14:19 2018 daemon.notice miniupnpd[10726]: Listening for NAT-PMP/PCP traffic on port 5351
Thu Dec 20 18:14:19 2018 daemon.err miniupnpd[10726]: PCPSendUnsolicitedAnnounce() IPv6 sendto(): Bad file descriptor

dnwk, 2018-12-21:
For now, I flashed back to v0.42.7 and works fine so far.

Ysurac, 2018-12-21:
Nothing special in log. Your problem will maybe be fixed with next Linux kernel...

PxxBxx, 2018-12-30:
Looking forward to future release (kernel?) because now that I monitor OMR, it reboot itself every couple of days.
I tried having logs in /tmp recently, but not log enough to catch last reboot. I'll watch for next one.

[image]

0.44.5 / X86_64 / 0.82

cron410, 2018-12-30:
PxxBxx have you tried different hardware? I have no issues on a x86_64 router 0.44.5. My uptime is 4d 16h 49m

PxxBxx, 2018-12-30:
@cron410 I didn't try to change hardware, it's a VM (Proxmox) with a classic hardware : 
[image]
I have no clue, and @dnwk had the same behaviour (random reboots)

Ysurac, 2018-12-30:
I use Virtualbox and I don't have this issue.

PxxBxx, 2019-01-01:
... I put system.log in /tmp, it's not persistent :'(
I try putting it in /root (/overlay/upper/root) and wait for next reboot for traces.
Meanwhile, I changed CPU to only one core, and NIC to e1000 (instead of virtio)
"maybe" it will change something because it's really strange all these reboots :-/

[image]

PxxBxx, 2019-01-01:
Do anybody knows if it's possible to have kernel logs written somewhere persistent ? (not system logs, I have these)
It keeps rebooting... and system.log has nothing interesting just before the reboot
(except nlbw having troubles dumping stats the last 2 hours before reboot)

PxxBxx, 2019-01-04:
Note : after rollbacking to a snapshot before installing nlbwmon, my uptime is improving.
Maybe nlbwmon is too aggressive in its conntrack/netfilter usage...
If anybody knows of any kernel tuning (sysctl) that can improve its stability, but it's not OMR-related, so i'll try with the nlbwmon community.

PxxBxx, 2019-01-11:
Some feedback : 
* 0.44.5 -> stable, constant uptime
* 0.44.5 + nlbwmon -> unstable, OMR rebooted itself every 2-4 hours after 5 days of bandwidth monitoring
* 0.45.6 + nlbwmon -> so far so good, 12h uptime (I'm joking, too early to know if it stable, but I'll let you know)

closed 2019-02-20 by Ysurac

cron410, 2019-03-26:
I've got 0.47.6 running in Proxmox for 21 hours now. I didn't try running nlbwmon yet. I'll let it hit a few days of uptime and then enable that to seeif there are more problems. This can stay closed until we find more issues with reboot.