Ysurac/openmptcprouter #271: Upgrade VPS not working

dnwk, 2018-12-21:
## Actual Behavior
when I click on System ->OpenMPTCProuter -> Advanced Settings ->Update VPS 
I got this error message. The same behavior has been there for 3 release now.

Failed to execute call dispatcher target for entry '/admin/system/openmptcprouter/update_vps'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/openmptcprouter.lua:509: attempt to index global 'uci' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/openmptcprouter.lua:509: in function 'callback'
	/usr/lib/lua/luci/model/uci.lua:295: in function 'foreach'
	/usr/lib/lua/luci/controller/openmptcprouter.lua:506: in function </usr/lib/lua/luci/controller/openmptcprouter.lua:502>
	(tail call): ?

## Specifications

  - OpenMPTCProuter version: (Last is not a version)   v0.44.5 r7416-50789acf4e
  - OpenMPTCProuter VPS version: (Last is not a version) 0.79
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) x86_64

Ysurac, 2018-12-21:
This will be fixed in next release.

Ysurac, 2019-01-06:
Fixed in latest release.

closed 2019-01-06 by Ysurac