Ysurac/openmptcprouter #273: Perte de connexion

Roken120, 2018-12-23:
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps=2 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps=1 ms TTL=64
Délai d’attente de la demande dépassé.
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Réponse de 192.168.0.33 : octets=32 temps<1ms TTL=64
Délai d’attente de la demande dépassé.
Délai d’attente de la demande dépassé.

J'ai des perte de connexion intempestive avec la box alors que quand ping avec le pc avec la box j'ai pas de perte.

Roken120, 2018-12-23:
Et je suis en ethernet.

Roken120, 2018-12-23:
le pc est sous windows

Ysurac, 2018-12-23:
La plateforme utilisé pour OpenMPTCProuter (RPI3B+, RPI2, machine virtuelle virtualbox, ...)

Ysurac, 2018-12-23:
Il faut voir dans les journaux systéme et du noyau si il n'y a rien de spécial.
Sinon il faut voir si au niveau du PC client il n'y a rien qui peut bloquer le ping, des routes étranges,...

Roken120, 2018-12-23:
Quand je me connecter sur les autres box je n'ai aucun problème c'est juste avec openmptcprouter.

Roken120, 2018-12-23:
Sun Dec 23 15:42:18 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:18 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:18 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:18 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:18 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.167 3c:6a:a7:20:91:7b DESKTOP-1MO3OO6
Sun Dec 23 15:42:19 2018 daemon.err glorytun[5175]: 154.49.213.44.65001: connected
Sun Dec 23 15:42:19 2018 daemon.info glorytun[5175]: STARTED tun0
Sun Dec 23 15:42:24 2018 daemon.info dnsmasq-dhcp[7695]: DHCPRELEASE(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:26 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:26 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:26 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.167 3c:6a:a7:20:91:7b
Sun Dec 23 15:42:26 2018 daemon.info dnsmasq-dhcp[7695]: DHCPNAK(lan) 192.168.0.167 3c:6a:a7:20:91:7b wrong server-ID
Sun Dec 23 15:42:35 2018 daemon.info dnsmasq-dhcp[7695]: DHCPINFORM(lan) 192.168.0.227 20:cf:30:ab:43:3a
Sun Dec 23 15:42:35 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.227 20:cf:30:ab:43:3a ucak-PC
Sun Dec 23 15:44:24 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 8e:e9:68:4c:24:d5
Sun Dec 23 15:44:24 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.164 8e:e9:68:4c:24:d5
Sun Dec 23 15:44:25 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.164 8e:e9:68:4c:24:d5
Sun Dec 23 15:44:25 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.164 8e:e9:68:4c:24:d5 OverTheBox
Sun Dec 23 15:44:30 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.175 a4:08:f5:c9:78:f5
Sun Dec 23 15:44:30 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.175 a4:08:f5:c9:78:f5
Sun Dec 23 15:44:36 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.195 54:bf:64:3e:c5:af
Sun Dec 23 15:44:36 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.195 54:bf:64:3e:c5:af DESKTOP-1MO3OO6
Sun Dec 23 15:47:14 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) d0:fc:cc:17:ed:1e
Sun Dec 23 15:47:14 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.168 d0:fc:cc:17:ed:1e
Sun Dec 23 15:47:14 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) d0:fc:cc:17:ed:1e
Sun Dec 23 15:47:14 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.168 d0:fc:cc:17:ed:1e
Sun Dec 23 15:47:14 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.168 d0:fc:cc:17:ed:1e
Sun Dec 23 15:47:14 2018 daemon.info dnsmasq-dhcp[7695]: DHCPNAK(lan) 192.168.0.168 d0:fc:cc:17:ed:1e wrong server-ID
Sun Dec 23 15:47:24 2018 daemon.err uhttpd[4229]: luci: accepted login on / for root from 192.168.1.123
Sun Dec 23 15:53:04 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.161 f0:18:98:22:55:ba
Sun Dec 23 15:53:04 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.161 f0:18:98:22:55:ba MBP-de-Enes
Sun Dec 23 15:58:30 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 6c:96:cf:92:34:19
Sun Dec 23 15:58:30 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.192 6c:96:cf:92:34:19
Sun Dec 23 15:58:30 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 6c:96:cf:92:34:19
Sun Dec 23 15:58:30 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.192 6c:96:cf:92:34:19
Sun Dec 23 15:58:31 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.192 6c:96:cf:92:34:19
Sun Dec 23 15:58:31 2018 daemon.info dnsmasq-dhcp[7695]: DHCPNAK(lan) 192.168.0.192 6c:96:cf:92:34:19 wrong server-ID
Sun Dec 23 16:00:00 2018 cron.info crond[3413]: USER root pid 14243 cmd /etc/init.d/vnstat_backup backup
Sun Dec 23 16:00:00 2018 user.info vnstat_backup: backing up database
Sun Dec 23 16:01:38 2018 daemon.info dnsmasq-dhcp[7695]: DHCPINFORM(lan) 192.168.0.227 20:cf:30:ab:43:3a
Sun Dec 23 16:01:38 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.227 20:cf:30:ab:43:3a ucak-PC
Sun Dec 23 16:16:56 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.192 6c:96:cf:92:34:19
Sun Dec 23 16:16:56 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.192 6c:96:cf:92:34:19 iPhone-derkan
Sun Dec 23 16:19:21 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:21 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.183 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:21 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:21 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.183 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:21 2018 daemon.info dnsmasq-dhcp[7695]: DHCPDISCOVER(lan) 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:21 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.183 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:24 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.183 6c:c7:ec:39:74:ab
Sun Dec 23 16:19:24 2018 daemon.info dnsmasq-dhcp[7695]: DHCPNAK(lan) 192.168.0.183 6c:c7:ec:39:74:ab wrong server-ID
Sun Dec 23 16:19:24 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.183 6c:c7:ec:39:74:ab
[... cut ...]
Sun Dec 23 19:03:07 2018 daemon.info dnsmasq-dhcp[7695]: DHCPOFFER(lan) 192.168.0.195 54:bf:64:3e:c5:af
Sun Dec 23 19:03:07 2018 daemon.warn dnsmasq-dhcp[7695]: DHCP packet received on eth0 which has no address
Sun Dec 23 19:03:07 2018 daemon.info dnsmasq-dhcp[7695]: DHCPREQUEST(lan) 192.168.0.195 54:bf:64:3e:c5:af
Sun Dec 23 19:03:07 2018 daemon.info dnsmasq-dhcp[7695]: DHCPACK(lan) 192.168.0.195 54:bf:64:3e:c5:af DESKTOP-1MO3OO6
Sun Dec 23 19:03:18 2018 daemon.err /usr/bin/ss-local[6405]: getpeername: Socket not connected

Roken120, 2018-12-23:
Voila les logs

Roken120, 2018-12-23:
Il y mo overthebox mais je veut l'enlever et le remplacer par l'open donc je deconnecter loverthebox je change l'ip de l'open en celui de l'overthebox (192.168.0.33)

Roken120, 2018-12-23:
J'ai desactiver le serveur de otb mais le même soucis
Sa me fait pareil

Roken120, 2018-12-23:
Tu peut m'apeller sur mon telephone au 06 21 86 33 28 pour m'aider stp car la c'est vraiement bizzare avecc l'otb aucun bug tous marche parfaitement avec l'open mes machines ping plus et des déconnexion intempestive

Ysurac, 2018-12-23:
Non je ne fais pas de support par tél.
* Si il y a toujours des soucis de DHCP, le conflit peut durer un peu, car une même adresse a pu être attribué à plusieurs machine (même si ça ne doit pas arriver).
* Si il n'y a plus de soucis de DHCP, il faut encore les logs.

Roken120, 2018-12-23:
[    0.000000] Linux version 4.14.88 (circleci@default-32583d86-b153-4b46-8164-b8d789ff5e0f) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7398-9671a2d2c8)) #0 SMP Sat Dec 15 17:26:49 2018
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=fb623262-02 rootfstype=squashfs rootwait console=tty0 noinitrd
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000095bff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000095c00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000cf65efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cf65f000-0x00000000cf67efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000cf67f000-0x00000000cf76efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cf76f000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed13fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed18000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000012bffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Dell Inc. Latitude E6410/, BIOS A07 02/15/2011
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x12c000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 base 080000000 mask FC0000000 write-back
[    0.000000]   2 base 0C0000000 mask FF0000000 write-back
[    0.000000]   3 base 100000000 mask FE0000000 write-back
[    0.000000]   4 base 120000000 mask FF0000000 write-back
[    0.000000]   5 base 12C000000 mask FFC000000 uncachable
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] e820: update [mem 0xd0000000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xcf65f max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000f2220-0x000f222f] mapped at [ffffffffff400220]
[    0.000000] Base memory trampoline at [ffff88800008f000] 8f000 size 24576
[    0.000000] BRK [0x02009000, 0x02009fff] PGTABLE
[    0.000000] BRK [0x0200a000, 0x0200afff] PGTABLE
[    0.000000] BRK [0x0200b000, 0x0200bfff] PGTABLE
[    0.000000] BRK [0x0200c000, 0x0200cfff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FE300 000024 (v02 DELL  )
[    0.000000] ACPI: XSDT 0x00000000CF67DE18 000064 (v01 DELL   E2       06222004 MSFT 00010013)
[    0.000000] ACPI: FACP 0x00000000CF75FC18 0000F4 (v04 DELL   E2       06222004 MSFT 00010013)
[    0.000000] ACPI: DSDT 0x00000000CF73E018 00A24E (v01 DELL   E2       00001001 INTL 20080729)
[    0.000000] ACPI: FACS 0x00000000CF76BF40 000040
[    0.000000] ACPI: FACS 0x00000000CF76ED40 000040
[    0.000000] ACPI: APIC 0x00000000CF67CF18 00008C (v02 DELL   E2       06222004 MSFT 00010013)
[    0.000000] ACPI: TCPA 0x00000000CF76DD18 000032 (v02                 00000000      00000000)
[    0.000000] ACPI: MCFG 0x00000000CF76DC98 00003C (v01 A M I  GMCH945. 06222004 MSFT 00000097)
[    0.000000] ACPI: HPET 0x00000000CF76DC18 000038 (v01 DELL   E2       00000001 ASL  00000061)
[    0.000000] ACPI: BOOT 0x00000000CF76DB98 000028 (v01 DELL   E2       06222004 AMI  00010013)
[    0.000000] ACPI: SLIC 0x00000000CF766818 000176 (v03 DELL   E2       06222004 MSFT 00010013)
[    0.000000] ACPI: SSDT 0x00000000CF74D018 0009F1 (v01 PmRef  CpuPm    00003000 INTL 20080729)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.000000]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000012bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[... 777 lines cut ...]
[  668.165904]  ? lapic_next_event+0x1e/0x30
[  668.166032]  ? clockevents_program_event+0xc0/0xe0
[  668.166159]  __do_softirq+0xba/0x1e5
[  668.166286]  irq_exit+0x62/0x90
[  668.166411]  smp_apic_timer_interrupt+0x87/0xa0
[  668.166536]  apic_timer_interrupt+0x81/0x90
[  668.166659]  </IRQ>
[  668.166782] RIP: 0010:cpuidle_enter_state+0x143/0x1c0
[  668.166908] RSP: 0018:ffffc9000006fe98 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff10
[  668.167090] RAX: ffff88812bc5f180 RBX: 0000000000000004 RCX: 000000000000001f
[  668.167217] RDX: 0000009b916d5724 RSI: 00000000301ec9a8 RDI: 0000000000000000
[... cut ...]
[  668.168342]  start_secondary+0x1a0/0x1b0
[  668.168467]  secondary_startup_64+0xa5/0xb0
[  668.168589] Code: 00 00 39 c3 75 c0 eb 41 4c 89 e7 c6 05 be 9e 98 00 01 e8 b1 14 fe ff 89 d9 48 89 c2 4c 89 e6 48 c7 c7 58 3b bd 81 e8 09 04 bc ff <0f> 0b 49 8b 84 24 f0 01 00 00 4c 89 e7 48 8b 40 78 e8 b7 d0 30 
[  668.168874] ---[ end trace 71520be3ce19c9c5 ]---
[  668.169007] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
[  671.371625] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx

Roken120, 2018-12-23:
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.399738] NET: Registered protocol family 10
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.401418] Segment Routing with IPv6
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.401600] fullmesh registered
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.401739] ndiffports registered
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.401877] binder registered
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.402015] roundrobin registered
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.402150] redundant registered
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.402285] NET: Registered protocol family 17
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.402437] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.402656] 8021q: 802.1Q VLAN Support v1.8
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.402880] NET: Registered protocol family 40
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.403548] microcode: sig=0x20655, pf=0x10, revision=0x2
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.403747] microcode: Microcode Update Driver: v2.2.
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.403758] SSE version of gcm_enc/dec engaged.
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.404923] sched_clock: Marking stable (404919942, 0)->(522533321, -117613379)
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.405201] registered taskstats version 1
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.405729] rtc_cmos 00:01: setting system clock to 2018-12-23 19:24:36 UTC (1545593076)
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.652645] ata6: SATA link down (SStatus 0 SControl 300)
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.652818] ata5: SATA link down (SStatus 0 SControl 300)
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.652993] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
Sun Dec 23 19:24:44 2018 kern.warn kernel: [    0.654730] ata1.00: supports DRM functions and may not be fully accessible
Sun Dec 23 19:24:44 2018 kern.warn kernel: [    0.655401] ata1.00: NCQ Send/Recv Log not supported
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.655548] ata1.00: ATA-9: Samsung SSD 850 EVO 250GB, EMT01B6Q, max UDMA/133
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.655696] ata1.00: 488397168 sectors, multi 1: LBA48 NCQ (depth 31/32), AA
Sun Dec 23 19:24:44 2018 kern.warn kernel: [    0.657224] ata1.00: supports DRM functions and may not be fully accessible
Sun Dec 23 19:24:44 2018 kern.warn kernel: [    0.657805] ata1.00: NCQ Send/Recv Log not supported
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.659021] ata1.00: configured for UDMA/133
Sun Dec 23 19:24:44 2018 kern.notice kernel: [    0.659378] scsi 0:0:0:0: Direct-Access     ATA      Samsung SSD 850  1B6Q PQ: 0 ANSI: 5
Sun Dec 23 19:24:44 2018 kern.notice kernel: [    0.660186] sd 0:0:0:0: [sda] 488397168 512-byte logical blocks: (250 GB/233 GiB)
Sun Dec 23 19:24:44 2018 kern.notice kernel: [    0.660458] sd 0:0:0:0: [sda] Write Protect is off
Sun Dec 23 19:24:44 2018 kern.debug kernel: [    0.660606] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
Sun Dec 23 19:24:44 2018 kern.notice kernel: [    0.660688] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.662133]  sda: sda1 sda2
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.662696] ata2: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
Sun Dec 23 19:24:44 2018 kern.notice kernel: [    0.663093] sd 0:0:0:0: [sda] Attached SCSI disk
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.668281] ata2.00: ATAPI: TEAC DVD-ROM DV18SA, D.0D, max UDMA/100
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.671944] ata2.00: configured for UDMA/100
Sun Dec 23 19:24:44 2018 kern.notice kernel: [    0.687477] scsi 1:0:0:0: CD-ROM            TEAC     DVD-ROM DV18SA   D.0D PQ: 0 ANSI: 5
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.730093] usb 1-1: new high-speed USB device number 2 using ehci-pci
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.760065] usb 2-1: new high-speed USB device number 2 using ehci-pci
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.767130] VFS: Mounted root (squashfs filesystem) readonly on device 8:2.
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.768678] devtmpfs: mounted
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.770202] Freeing unused kernel memory: 1056K
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.800184] Write protecting the kernel read-only data: 14336k
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.801247] Freeing unused kernel memory: 2024K
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.810747] Freeing unused kernel memory: 1884K
Sun Dec 23 19:24:44 2018 user.info kernel: [    0.887971] init: Console is alive
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.930751] hub 1-1:1.0: USB hub found
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.930956] hub 1-1:1.0: 6 ports detected
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.960572] hub 2-1:1.0: USB hub found
Sun Dec 23 19:24:44 2018 kern.info kernel: [    0.960812] hub 2-1:1.0: 8 ports detected
Sun Dec 23 19:24:44 2018 user.info kernel: [    1.191506] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Sun Dec 23 19:24:44 2018 kern.info kernel: [    1.201499] pps_core: LinuxPPS API ver. 1 registered
[... cut ...]
Sun Dec 23 19:47:29 2018 daemon.err /usr/bin/ss-redir[6456]: remote recv: Connection reset by peer
Sun Dec 23 19:47:29 2018 daemon.err /usr/bin/ss-redir[6458]: remote recv: Connection reset by peer
Sun Dec 23 19:47:38 2018 daemon.err /usr/bin/ss-redir[6459]: remote recv: Connection reset by peer
Sun Dec 23 19:48:07 2018 daemon.err /usr/bin/ss-redir[6462]: remote recv: Connection reset by peer
Sun Dec 23 19:48:16 2018 daemon.err /usr/bin/ss-redir[6460]: remote recv: Connection reset by peer
Sun Dec 23 19:48:32 2018 daemon.err /usr/bin/ss-redir[6458]: remote recv: Connection reset by peer

Ysurac, 2018-12-23:
Le soucis : ```e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang```.
Ca peut être un soucis de câble, de switch, de matériel...
On peut essayer de désactiver les checksums sur la carte ```ethtool -K eth0 tx off rx off``` en SSH sur OpenMPTCProuter.

Roken120, 2018-12-23:
Toujours pareil le même problème jai changer de machine sa fait le même bug

Roken120, 2018-12-23:
Oui jai même changer de machine et mis celui ou javais lotb mais toujours pareil

Ysurac, 2018-12-23:
Je peux pas dire grand chose de plus. Un soucis matériel est détecté par le noyau.

closed 2019-02-20 by Ysurac