Ysurac/openmptcprouter #277: new version not working

kapook007, 2018-12-26:
hello
my system  is x86 cpu.
OpenMPTCProuter v.0.38  and  VPS v0.64   is  working successful.

i flash OpenMPTCProuter to v.044 and  VPS v0.77 have some problem.

1. OpenMPTCProuter can not ping glorytun vps gateway ( 10.255.255.1)
2. Network overview  wan1-wan5 show "multipath seems to be blocked on the connection"

but only tcp  shadowsocks  can run  successful.
my key shadowsocks  and   glorytun is correct.

Ysurac, 2018-12-26:
Latest version are OpenMPTCProuter 0.44.5 and OpenMPTCProuter VPS 0.80. You should update.

kapook007, 2018-12-26:
WOW  thank you

kapook007, 2018-12-28:
OpenMPTCProuter 0.44.5 and OpenMPTCProuter VPS 0.80. 
smae problem
OpenMPTCProuter can not ping glorytun vps gateway ( 10.255.255.1)

Ysurac, 2018-12-28:
You should check OpenMPTCProuter log.

kapook007, 2018-12-28:
my OpenMPTCProuter VPS  is behind nat
now i forward port 65101 and 65001

how many  rang forward port ?

Ysurac, 2018-12-28:
All ports used are here: https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

closed 2019-01-06 by Ysurac