Ysurac/openmptcprouter #278: enhancement : monitor each client of the gateway

PxxBxx, 2018-12-28:
I'm putting the idea here, it would help me discover any clients "leeching" bandwidth (who said Windows 10 peering ?? nvidia shield loop updates ?? etc.).

=> to have a "per client" monitoring of the gateway usage
=> nice to have : QoS per device

I really had a culprit between 2am and 6am, it was the nVidia Shield updating, many times for the same upgrade.... a reset corrected it, and here is how I found out : 
For the monitoring part, I tried https://github.com/Kiougar/luci-wrtbwmon, and the interface is interesting, but it only works when the page is open and updating itself.
So I added a CRON to update the usage.db file every minutes (a simple CSV file)
``* * * * * /usr/sbin/wrtbwmon update /etc/config/usage.db``
Then I tried to monitor the values (via Grafana), and everything is not catched : 
 - overall, all bytes are counted, total seems close to SNMP interfaces values.
 - but sometimes when 2 devices are on netflix or youtube, bytes are counted on only one client (double value) not both, maybe a iptable conflict (due to bypass ? I don't know...). So i'm not totally confident in this solution (but iptables counters are the way, I think.

I start this thread, because maybe someone with a better WRT background know of a good plugin to test, before maybe integrating it into OMR ?

Requirements : 
 - monitor throughput for each client of the gateway
 - luci interface to easily see the datas/activities/... of the clients
 - snmp or easily exportable datas (target : cacti spine or influx telegraf or else)

Do you think it would be a good feature for OMR ?

Ysurac, 2018-12-28:
You can try *luci-app-nlbwmon* package.

PxxBxx, 2018-12-29:
Thanks a lot !
I searched like hell to install nlbwmon..... no way using the "software" page in luci interface, I was looking how to build it before finding the right way : 
`opkg update && opkg install luci-app-nlbwmon`

I will let it run some days to see if it's worth it. Right now, it seems great !

closed 2019-01-27 by Ysurac