Ysurac/openmptcprouter #279: Can't access and use OpenMPTCServer server

YUjji, 2018-12-29:
New VPS installation done on a Vmware with a pfsense as a firewall
the ports reminding sour towards the VM:
65222 SSH
65101 Shadowsocks
65001 Glorytun
65500 OMR
65301 OpenVPN
65201 - 65208 MLVPN

I end up with an error message
Can not access and use OpenMPTCServer server

And I have the VPN tunnel that falls from time to time which makes me mounted my ping from 30ms to 500ms

Do you have a solution ?

## Specifications

  - OpenMPTCProuter version: 0.44.5
  - OpenMPTCProuter VPS version: 0.80
  - OpenMPTCProuter platform: RPI3

Ysurac, 2018-12-29:
Do you follow https://github.com/Ysurac/openmptcprouter/wiki/Router-install#for-esxi ?

YUjji, 2018-12-29:
The VPS is on a Vmware the router is an RPI3
It worked well before I put all the news today

YUjji, 2018-12-30:
Thanks the ping looks more stable I will check during the day and I would make a return against the problem of connection to the server that can come from where ?

[image]

closed 2019-01-06 by Ysurac