Ysurac/openmptcprouter #282: Constantly 'VPN tunnel DOWN' or 'Can't contact Server Admin Script'

max4186, 2018-12-30:
Ever since the latest VPS update, 0.82 - the router has been EXTREMELY buggy. For some reason, my setup keeps switching between 'VPN tunnel DOWN' and then everything will go back online and then it'll switch to being online but 'Can't contact Server Admin Script' when I am currently connected fine without any issues.

It's getting rather annoying, I've done a full reinstall of the VPS and done a fresh install and it has changed nothing. I also flashed new firmware onto the RPI 3 and have had no luck. I have attached the only kind of logs I can really find

`Sun Dec 30 05:37:25 2018 daemon.info glorytun[7580]: STOPPED tun0
Sun Dec 30 05:37:26 2018 daemon.err glorytun[7580]: 185.234.74.6.65001: connected
Sun Dec 30 05:37:26 2018 daemon.info glorytun[7580]: STARTED tun0
Sun Dec 30 05:37:52 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:37:53 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:37:53 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Dec 30 05:38:14 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:38:14 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:38:17 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Dec 30 05:38:34 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:38:34 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:38:36 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Dec 30 05:38:54 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:38:54 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:38:57 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Dec 30 05:39:11 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:39:11 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:39:14 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Dec 30 05:39:30 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:39:31 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:39:36 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Dec 30 05:39:50 2018 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Dec 30 05:39:51 2018 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Sun Dec 30 05:39:51 2018 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0`

Is anyone else having these problems with the latest update?

max4186, 2018-12-30:
As of writing this, it's not sticking to solid but I'm going to go to sleep and won't be surprised if it's back to 'VPN tunnel DOWN' - please note that I switched between GloryTUN and MLVPN to test and no difference.

max4186, 2019-01-01:
I have updated, the issue is still occurring.

https://i.cyrex.io/HeLNAWO.png

max4186, 2019-01-01:
@Ysurac

Ysurac, 2019-01-01:
Glorytun seems to restart every few seconds on your RPI... I think you have an error in your configuration.

max4186, 2019-01-01:
> Glorytun seems to restart every few seconds on your RPI... I think you have an error in your configuration.

It's the exact same when using MLVPN

https://i.cyrex.io/6eTN7C1.png - GloryTUN off
https://i.cyrex.io/he6m3bw.png - MLVPN on

I can't find any errors at all other than this which seems to be spammed but hasn't done in over 40 minutes - please note this has only started since latest updates and has had the EXACT same configuration for the past 3 weeks without this problem.

ue Jan  1 05:54:05 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:54:13 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:54:35 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:54:44 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:55:05 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:55:14 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:55:35 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:55:44 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:56:05 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:56:14 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:56:35 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:56:44 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:57:05 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:57:14 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:57:35 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:57:44 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:58:05 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:58:14 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:58:36 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:58:44 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:59:06 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:59:14 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:59:36 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 05:59:44 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:00:00 2019 cron.info crond[1667]: USER root pid 9135 cmd /etc/init.d/vnstat_backup backup
Tue Jan  1 06:00:00 2019 user.info vnstat_backup: backing up database
Tue Jan  1 06:00:06 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:00:14 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:00:36 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:00:45 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:01:06 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:01:15 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:01:36 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:01:45 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:02:06 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:02:15 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:02:36 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:02:45 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:03:06 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:03:15 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected
Tue Jan  1 06:03:36 2019 daemon.err /usr/bin/ss-redir[4294]: crypto: stream: repeat IV detected

Ysurac, 2019-01-01:
You restarted the VPS after the update ?

max4186, 2019-01-01:
> You restarted the VPS after the update ?

Yep I have, as well as the RPI and actually a FULL reinstall of the VPS as well as re-flashing firmware onto the RPI through the built in flasher on the panel.

Is there any specific commands I can run or tests I can do to make sure the VPN actually is running to rule that out?

Ysurac, 2019-01-01:
When you change the VPN used, you should always use the wizard.
You really should try to configure the RPI from crash if you used a previous configuration.

You can add up to 8 modems. If it's ethernet, use the wizard, if it's USB you have to add it via network->interface config.