Ysurac/openmptcprouter #283: Issue Port Forward

bran2000, 2018-12-30:
Hi,

Firstable, thanks for your work ! it's amazing !
Last thing, i met a problem to access outside to my synology nas with my public ip : 5x.XX.XX.XX and the port 5000

When i did this with my browser : http:\\5x.XX.XX.XX:5000 it's not working, i have the message time out.

i followed your tutorial but it's not working 

  1. Forward to	Enable	
IPv4-tcp
From any host in vpn
Via any router IP at port 5000	
IP 192.168.100.57, port 5000 in lan	
Enable NAT Loopback (untick)

  2. Sources NAT
Any tcp
From any host in vpn
To IP 192.168.100.57, port 5000 in lan	
Rewrite to source IP 192.168.100.1	

any idea ? i guess i missed something but i don't know where ?

Thanks

## Specifications

  - OpenMPTCProuter version: v0.44.5
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.81
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2018-12-30:
You redirected all ports from VPS to OpenMPTCProuter ? (Use the checkbox in System->OpenMPTCProuter then Advanced settings tab.)

bran2000, 2018-12-30:
thanks i guess it was this i tick this field : Redirects all ports from VPS to OpenMPTCProuter
one last question : if i want to access to the admin console Openmptcprouteur outside , i guess i shoud do the same but with the port 80 ?

Ysurac, 2018-12-30:
True, you have to accept input traffic from VPN to port 80 of OpenMPTCProuter in firewall rules.

bran2000, 2018-12-30:
thanks for your quick reply, can i activate a login/password to access to admin console ?

Ysurac, 2018-12-30:
Yes in System->Admin

bran2000, 2018-12-30:
great Thanks for your help again !

closed 2018-12-30 by Ysurac