Ysurac/openmptcprouter #284: Version display in router status

trassiorf, 2018-12-30:
Hi 
Why I have a VPS version on 0.82 after update VPS while status display last version is 0.83 ?
See screenshot 
[image]

## Specifications

  - OpenMPTCProuter version: (Last is not a version) 0.44.5
  - OpenMPTCProuter VPS version: (Last is not a version) 0.82
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3 B+

Ysurac, 2018-12-30:
Because latest version is 0.83 now :)

trassiorf, 2018-12-30:
Launching the script : # sh debian9-x86_64.sh at 15:52 this day
I stay on version 0.82

Ysurac, 2018-12-30:
On the VPS you should have version 0.83, even if status is not updated every minutes...

trassiorf, 2018-12-30:
Well, I did not update the script.
Now the version is 0.83 after the last compilation phase.
Thank you

closed 2018-12-30 by trassiorf