Ysurac/openmptcprouter #286: Disconnection of the ethernet card

teraska47, 2018-12-31:
## Expected Behavior
In 0.80 of the VPS no disconnection of the Ethernet card

## Actual Behavior

since my switch to 0.83 VPS mode I constantly lose the connection to the VPN following the loss of the ethernet card.
here are the logs  
## Steps to Reproduce the Problem

  1.upgrade of the VPS from 0.80 to 0.83
  2.
  3.

## Specifications

  - OpenMPTCProuter version: 0.44.5
  - OpenMPTCProuter VPS version: 0.83
  - OpenMPTCProuter platform: x86_64

The logs systems
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang:
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   TDH                  <4c>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   TDT                  <85>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   next_to_use          <85>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   next_to_clean        <3a>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] buffer_info[next_to_clean]:
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   time_stamp           <1003cb311>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   next_to_watch        <4c>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   jiffies              <1003cb380>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606]   next_to_watch.status <0>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] MAC Status             <80083>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] PHY Status             <796d>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] PHY 1000BASE-T Status  <3c00>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] PHY Extended Status    <3000>
Mon Dec 31 10:48:48 2018 kern.err kernel: [40080.119606] PCI Status             <10>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang:
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   TDH                  <4c>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   TDT                  <85>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   next_to_use          <85>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   next_to_clean        <3a>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] buffer_info[next_to_clean]:
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   time_stamp           <1003cb311>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   next_to_watch        <4c>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   jiffies              <1003cb448>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758]   next_to_watch.status <0>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] MAC Status             <80083>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] PHY Status             <796d>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] PHY 1000BASE-T Status  <3c00>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] PHY Extended Status    <3000>
Mon Dec 31 10:48:50 2018 kern.err kernel: [40082.119758] PCI Status             <10>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang:
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   TDH                  <4c>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   TDT                  <85>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   next_to_use          <85>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   next_to_clean        <3a>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] buffer_info[next_to_clean]:
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   time_stamp           <1003cb311>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   next_to_watch        <4c>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   jiffies              <1003cb510>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475]   next_to_watch.status <0>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] MAC Status             <80083>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] PHY Status             <796d>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] PHY 1000BASE-T Status  <3c00>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] PHY Extended Status    <3000>
Mon Dec 31 10:48:52 2018 kern.err kernel: [40084.119475] PCI Status             <10>
Mon Dec 31 10:48:54 2018 daemon.notice netifd: Network device 'eth0' link is down
Mon Dec 31 10:48:54 2018 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Dec 31 10:48:54 2018 daemon.notice netifd: macvlan 'wan1' link is down
Mon Dec 31 10:48:54 2018 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Mon Dec 31 10:48:54 2018 daemon.notice netifd: macvlan 'wan2' link is down
[... 254 lines cut ...]
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518]   jiffies              <1003cf908>
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518]   next_to_watch.status <0>
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518] MAC Status             <80083>
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518] PHY Status             <796d>
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518] PHY 1000BASE-T Status  <3c00>
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518] PHY Extended Status    <3000>
Mon Dec 31 10:51:46 2018 kern.err kernel: [40258.117518] PCI Status             <10>
Mon Dec 31 10:51:48 2018 daemon.notice netifd: Network device 'eth0' link is down
Mon Dec 31 10:51:48 2018 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Dec 31 10:51:48 2018 daemon.notice netifd: macvlan 'wan1' link is down
Mon Dec 31 10:51:48 2018 daemon.notice netifd: Interface 'wan1' has link connectivity loss
Mon Dec 31 10:51:48 2018 daemon.notice netifd: macvlan 'wan2' link is down
Mon Dec 31 10:51:48 2018 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Mon Dec 31 10:51:48 2018 kern.err kernel: [40260.117143] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
Mon Dec 31 10:51:52 2018 daemon.notice netifd: Network device 'eth0' link is up
Mon Dec 31 10:51:52 2018 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Dec 31 10:51:52 2018 daemon.notice netifd: macvlan 'wan1' link is up
Mon Dec 31 10:51:52 2018 daemon.notice netifd: Interface 'wan1' has link connectivity
Mon Dec 31 10:51:52 2018 daemon.notice netifd: macvlan 'wan2' link is up
Mon Dec 31 10:51:52 2018 daemon.notice netifd: Interface 'wan2' has link connectivity
Mon Dec 31 10:51:52 2018 kern.info kernel: [40263.939084] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
Mon Dec 31 10:52:30 2018 daemon.err glorytun[4867]: read: Operation timed out
Mon Dec 31 10:52:30 2018 daemon.info glorytun[4867]: STOPPED tun0
Mon Dec 31 10:52:31 2018 daemon.err glorytun[4867]: xxxxxxx.65001: connected
Mon Dec 31 10:52:31 2018 daemon.info glorytun[4867]: STARTED tun0

Ysurac, 2018-12-31:
The log come from the router, the VPS update only update the VPS part. It's not related.
You can try to disable TSO, GSOand GRO on OpenMPTCProuter: ```ethtool -K eth0 gso off gro off tso off```

teraska47, 2018-12-31:
Yes you're right it's a bug on the Intel e1000e card.
Yes, I'm here to see how to solve the problem, with your commands and with a script
http://e1000.sourceforge.net/files/fixeep-82573-dspd.sh

teraska47, 2018-12-31:
My hardware is a Ibm Lenovo ThinkCentre mini pc M73, 8G of Ram, processor Intel Pentium G3220T - 2.6 GHz

closed 2019-02-20 by Ysurac